PX4/PX4-Autopilot #13009: jMAVSim无法启动 ＃10377

love-mj, 2019-09-23:
[1/1] cd /home/mj/src/Firmware/build/px4_sitl_default/tmp .../src/Firmware /home/mj/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/mj/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/mj/src/Firmware
build_path: /home/mj/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/mj/src/Firmware/build/px4_sitl_default/bin/px4" "/home/mj/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/mj/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/mj/src/Firmware/ROMFS/px4fmu_common -> /home/mj/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
Buildfile: /home/mj/src/Firmware/Tools/jMAVSim/build.xml
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet

make_dirs:

compile:
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet

create_run_jar:
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet

copy_res:
ERROR [px4] PX4 daemon not running yet

BUILD SUCCESSFUL
Total time: 0 seconds
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
Error: no autostart file found (etc/init.d-posix/ERROR [px4] PX4 daemon not running yet_*)
ERROR [px4] Startup script returned with return value: 256
pxh> Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

julianoes, 2019-09-24:
Tried these steps?
https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting

love-mj, 2019-09-24:
I'll try.

julianoes, 2019-09-25:
Please re-open if the issue persists.

closed 2019-09-25 by julianoes