PX4/PX4-Autopilot #13012: SITL Hippocampus Sensor Baro #0 failoever

LukasAKeen, 2019-09-23:
I get the following error when attempting to `make px4_sitl gazebo_hippocampus`

[image]

I believe this is related to me not being able to use the `commander takeoff` function as it returns
`ERROR [commander] rejecting takeoff, no position lock yet. Please retry..`

julianoes, 2019-09-25:
I can reproduce this. There seems to be no GPS lat/lon/alt.

```
pxh> gpssim status
INFO  [gpssim] protocol: SIM
INFO  [gpssim] sat info: disabled, noise: 0, jamming detected: NO
 vehicle_gps_position_s
	timestamp: 79564000  (0.000000 seconds ago)
	time_utc_usec: 0
	lat: 0
	lon: 0
	alt: 0
	alt_ellipsoid: 0
	s_variance_m_s: 0.2500
	c_variance_rad: 0.0000
	eph: 655.3500
	epv: 655.3500
	hdop: 0.0000
	vdop: 0.0000
	noise_per_ms: 0
	jamming_indicator: 0
	vel_m_s: 0.0000
	vel_n_m_s: 0.0000
	vel_e_m_s: 0.0000
	vel_d_m_s: 0.0000
	cog_rad: 0.0000
	timestamp_time_relative: 0
	heading: nan
	heading_offset: nan
	fix_type: 0
	vel_ned_valid: False
	satellites_used: 0
```

julianoes, 2019-09-25:
@LukasAKeen I suggest you do a `git bisect` to figure out when it broke. That's assuming there is some version in the past where it still works.

ashishraste, 2019-12-05:
Hi @julianoes : I'm also facing this issue while trying to simulate Hippocampus UUV: barometer and magnetometer's simulated readings go stale on PX4's boot. 

Noticed the following:

* In debug mode their measurements do come through for few seconds before going stale i.e. with same timestamps.
* Tried changing the simulation `lockstep` by setting `PX4_SIM_SPEED_FACTOR` to 10 from the default value of 1 in debug mode without any breakpoints, and the issue persists.
* Control commands from the discussion on #7888 work with stale barometer and magnetometer measurements.

Please find below my console log from running `make px4_sitl_default gazebo_hippocampus`. 

Would like to know how this can be fixed so that a more complete controls-simulation test could be achieved.
 
Appreciate your help very much.

```
______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_1020
[param] Loaded: eeprom/parameters_1020
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.220
INFO  [init] Mixer: etc/mixers/uuv_quad_x.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
ERROR [sensors] Baro #0 fail:  STALE!
ERROR [sensors] Sensor Baro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Baro #0 priority: 1
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-12-05/08_47_28.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
ERROR [sensors] Mag #0 fail:  STALE!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1

Exiting...
pxh> Shutting down
```

julianoes, 2019-12-06:
Have you tried without lockstep?
https://dev.px4.io/master/en/simulation/#disable-lockstep-simulation

ashishraste, 2019-12-10:
Thanks for replying. Baro/Magnetometer error exists after disabling lockstep in Gazebo and SITL board.

julianoes, 2020-03-10:
> Baro/Magnetometer error exists after disabling lockstep in Gazebo and SITL board.

Only initially and it still let's you execute everything, right?

Jaeyoung-Lim, 2021-02-01:
This has been fixed on latest master, closing

closed 2021-02-01 by Jaeyoung-Lim