PX4/PX4-Autopilot #13013: FW 1.9.2 can't find ms5525 sensor

minhnguyne, 2019-09-23:
Hello,
I'm using QGroundControl (FW 1.9.2) to configure mRo Next-Gen MS5525 Airspeed sensor with the Pixracer r15, but it is giving me this error. 

[image]

I have already gone through this similar thread, but the same issue still occurs
[https://github.com/PX4/Firmware/issues/12375](url)
Any help would be greatly appreciated, 
Thank you!

mcsauder, 2019-09-25:
Hi @minhnguyne , I'm on the road this week, but I'll be able to test with a pixracer this weekend and get back to you with anything I learn.  If you figure anything out in the meantime let me know!

minhnguyne, 2019-09-25:
Okay, thank you!

minhnguyne, 2019-09-26:
It works now. I've rewired the MS5525 wires back into the Pixracer. I think the issue was that the SCL and SDA wires were swapped and/or unsecured. Thanks for your quick reply!

closed 2019-09-26 by minhnguyne