PX4/PX4-Autopilot #13019: RDDRONE-FMUK66 Parameters not being saved on some units

igalloway, 2019-09-24:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

davids5, 2019-09-25:
@igalloway Tested on master with REV c HW.  Params are AOK.

I can we do a postmortem on the device that is failing if you send it to me.

davids5, 2019-09-25:
@igalloway Oh by the way 75b2053243cb569c1347686fd56e33459e095efb removed uavcan by accident. @dagar will restore it.

igalloway, 2019-09-25:
Comment : @igalloway I'm on board #2 with the same issue. Were you able to find out if it's a manufacturing issue, and if so, can you provide any details? I'm hesitant to request another replacement before we get this figured out, especially if it's something that I can correct on my end. I get similar results as @physicsman on both boards. An SD card with dataman, a mission_log folder, and a log folder.

igalloway, 2019-09-25:
I'll try to get one of the failing board asap

igalloway, 2019-09-26:
Note: Confirmation by a 3rd party today that earlier code on the same board  **does not** fix the issue.

davids5, 2019-09-27:
That would indicate HW as does my test on Rev C. master 9/25 and QGC. (Prior I was just testing from command line. `set param xxxx yyyy'  and reboot check that is was still set.) Looking forward to receiving failing HW.

igalloway, 2019-10-07:
Thanks @davids5 for the investigation. It was determined that indeed this is a result of bad FRAM chips. We will investigate the source. 
The chips in question are marked correctly, but report back as a larger 64K part instead of the expected 16K.  This can be confirmed with `mtd erase` command. 
David created a temporary branch of code that also reported the JEDEC ID of the chip.
Possible causes:
 - bad chip/wear/old lot/ not passing qual/ not programmed internal mask correctly?
- grey market part by Contract manufacturer.

All current inventory will be tested. Our CM will be informed and an additional test at their facility will be implemented.
We will attempt to contact the manufacturer of this part for further diagnosis. This FRAM was not an NXP part.

closed 2019-10-07 by igalloway