PX4/PX4-Autopilot #13022: CI failure boost::lock_error

julianoes, 2019-09-25:
I have come across a new CI failure:

```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'

  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
```

It happened here:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13018/1/pipeline

The output is:
```

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [px4_work_queue] creating: wq:hp_default, priority: 87, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 88, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 99, stack: 16384 bytes
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14558 remote port 14541
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-09-24/16_45_35.ulg
INFO  [px4] Startup script returned successfully
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [ekf2] Mag sensor ID changed to 197388
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
INFO  [commander] Takeoff detected
WARN  [FlightTaskUtility] Obstacle Avoidance system failed, loitering

Exiting...
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
 78.860000000]: [WGN] Current position -0.001366 -0.010335 0.261985[0m
[0m[ INFO] [1569343629.894340536, 78.888000000]: [WGN] Current position -0.001168 -0.010231 0.262298[0m
[0m[ INFO] [1569343629.895243620, 78.888000000]: [WGN] Not a land waypoint[0m
[0m[ INFO] [1569343629.895284755, 78.888000000]: [1;32m [WGN] goTo -0.017780 -0.021857 0.284546 - nan nan nan [0m
[0m
[0m[ INFO] [1569343629.895311656, 78.888000000]: [WGN] Landing Radius: xy  0.020277, z 0.262298 [0m
[0m[ INFO] [1569343629.967231499, 78.960000000]: [WGN] Current position -0.000286 -0.009067 0.263172[0m
[0m[ INFO] [1569343629.994863449, 78.988000000]: [WGN] Current position 0.000027 -0.008655 0.263742[0m
[0m[ INFO] [1569343630.026450692, 79.016000000]: [WGN] Current position 0.000289 -0.008328 0.264379[0m
[0m[ INFO] [1569343630.027940583, 79.016000000]: [WGN] Not a land waypoint[0m
[0m[ INFO] [1569343630.068198662, 79.060000000]: [WGN] Current position 0.000512 -0.008083 0.264906[0m
[0m[ INFO] [1569343630.096034772, 79.084000000]: [WGN] Current position 0.000714 -0.007920 0.265384[0m
[0m[ INFO] [1569343630.127630980, 79.120000000]: [WGN] Current position 0.000918 -0.007317 0.265793[0m
[0m[ INFO] [1569343548.120150071, 0.380000000]: Physics dynamic reconfigure ready.[0m
[1;33m[0m[1;36m[0m[1;31m[0m[1;32m[0m88728, 19.076000000]: FCU: Takeoff detected[0m
[0m[ INFO] [1569343570.442748906, 20.428000000]: FCU: Avoidance system timed out[0m
[0m[ INFO] [1569343570.769348084, 20.740000000]: FCU: Avoidance system connected[0m
... logging to /tmp/jenkins/workspace/isc_Firmware-SITL_tests_PR-13018/.ros/ros_logs/rostest-9cccc34318f3-3916.log
[ROSUNIT] Outputting test results to /tmp/jenkins/workspace/isc_Firmware-SITL_tests_PR-13018/.ros/test_results/px4/rostest-test_mavros_posix_test_safe_landing.xml
[Testcase: testMC_safe_landing] ... ok

[ROSTEST]-----------------------------------------------------------------------

[px4.rosunit-MC_safe_landing/test_mission][FAILURE]-----------------------------
position not reached | lat: 47.397739400, lon: 8.545594200, alt: 506.03, current pos_xy_d: 0.27, current pos_z_d: 17.72, best pos_xy_d: 0.19, best pos_z_d: 16.62, index: 0 | timeout(seconds): 60
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/tmp/jenkins/workspace/isc_Firmware-SITL_tests_PR-13018/px4-px4_sitl_default-v1.10.0-beta1-123-gc435e45/px4/integrationtests/python_src/px4_it/mavros/mission_test.py", line 270, in test_mission
    index)
  File "/tmp/jenkins/workspace/isc_Firmware-SITL_tests_PR-13018/px4-px4_sitl_default-v1.10.0-beta1-123-gc435e45/px4/integrationtests/python_src/px4_it/mavros/mission_test.py", line 227, in reach_position
    best_pos_z_d, index, timeout))
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 1
 * ERRORS: 0
[1;31m * FAILURES: 1[0m

rostest log file is in /tmp/jenkins/workspace/isc_Firmware-SITL_tests_PR-13018/.ros/ros_logs/rostest-9cccc34318f3-3916.log
```

julianoes, 2019-09-25:
@mrivi have you seen this before?

mrivi, 2019-09-25:
yes, when we didn't run xvfb for the camera simulation. has anything changed on the containers side?
is the avoidance test failing as well?

mrivi, 2019-09-25:
@julianoes  running fine now http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13018/3/pipeline/

julianoes, 2019-09-25:
It seems to be just intermittent, not sure.

Jaeyoung-Lim, 2020-01-05:
Has there been any update on this? I can also see the same error when running the rostest locally

julianoes, 2020-02-12:
No idea what to do about it except ignore it :confused:.

julianoes, 2024-03-04:
Haven't seen this in a while.

closed 2024-03-04 by julianoes