PX4/PX4-Autopilot #13032: Triggering of "Autopilot off" by RC inputs

sfuhrer, 2019-09-26:
**Describe the bug**
For triggering the "autopilot off" the current RC inputs are compared to the last time the vehicle was in a manual mode (
Thus, when the RC sticks are not in neutral when the switch into mission happens, the failsafe is triggered and the warning appears. The vehicle though proceeds in it's mission.

On the other hand, if the RC sticks are really moved during the transition, then the mission is correctly aborted and control returned to the pilot.

**To Reproduce**
In sitl VTOL: 
- plan a mission
- take off in manual mode (e.g. position controlled)
- give some stick inputs and at the same time switch into mission mode
- the warning "Autopilot off. Return control to pilot" appears.
- the mission is still executed

**Expected behavior**
2 different things:
- I would expect that for detecting RC inputs, movement of the RC stick is detected, not the difference to where the sticks were whens the mission started
- The mission shouldn't  be continued/started when the "Autopilot off. Return control to pilot"-warning appeared

sfuhrer, 2019-09-26:
@julianoes could you have a look at this?

julianoes, 2019-09-30:
Ok thanks for reporting, that sounds like a bug. Now the question is if:
1. The behaviour should be changed, so we actually go back into manual mode.
2. Or the warning should not be printed but we tolerate that the sticks need to go into neutral position.

I would probably vote for 2. because it sounds like a real use case when going from fixedwing manual congtrol to an auto mode.

sfuhrer, 2019-10-01:
@julianoes I agree, also would prefer 2. Maybe we should only consider stick inputs that increase the absolute value of the stick i.e. abs(stick-value)

julianoes, 2019-11-20:
@sfuhrer you have fixed this, right?

sfuhrer, 2019-11-20:
No I haven't. @MaEtUgR was looking briefly into it if I remember correctly.

MaEtUgR, 2019-11-22:
I didn't yet. Could it be that the message is just delayed a bit? Otherwise it's a commander glitch depending on timing. Maybe because of low prio, high prio thread.