PX4/PX4-Autopilot #13035: VTOL mission can't be continued afte QuadChute

sfuhrer, 2019-09-27:
When a QuadChute is triggered, the vehicle immediately starts with a RTL. The mission can't be continued (also not when first switched into position control mode). 

Only Change Altitude action available:
[image]

Wouldn't it be better if after a Quadchute, the vehicle stays in hold mode (for a parameter-defined amount of time), before starting a RTL? And that the operator always has the option to continue the mission.

Further, I noticed different RC override behavior: In return mode after a QuadChute, RC override with moving the sticks works, whereas during a normal RTL (e.g. triggered by the RTL slider) one has to switch the flight mode (e.g. into position mode) to be able to take over. Is that so by design?

LorenzMeier, 2019-09-27:
On the RTL question: That is by design. Otherwise users abort RTL as the result of any failsafe action my fiddling around with the sticks without noticing. That way a low battery RTL gets aborted by an inexperienced user and they crash with empty batteries.

julianoes, 2019-10-01:
> That way a low battery RTL gets aborted by an inexperienced user and they crash with empty batteries.

Before crashing we still switch to landing though, right?

moreba1, 2019-10-04:
Vtol plane Stall cause QuadChute be acticvated . so i will to transition to FW again in RTL mode ,but i can not . in my opinion changing mode and transitions must be allowed in all conditions.

closed 2020-05-02 by sfuhrer