PX4/PX4-Autopilot #13054: S500 Airframe problems apart from a certain commit after 1.9 on

MaEtUgR, 2019-09-30:
**Describe the bug**
I didn't test/experience this myself but I've read now multiple times that people flying S500's mostly from Holybro have certain oscillations problems but only from a certain version after 1.9.2 on. 1.9.2 reproducably works well. I think we should check this with our own airframes before releasing 1.10 to the public.

**To Reproduce**
Here's one example: https://discuss.px4.io/t/default-s500-airframe-configuration-not-flyable-with-default/12394
There were others on Slack.

**Expected behavior**
The Airframe flies nice with the default tested components from Holybro.

**Drone:**
- Holybro S500

junwoo091400, 2022-11-22:
Oh noticed this too with X500 V2, as it was shaking a lot with default parameters 🤔 

But according to that user's post in the forum, it was fixed in PX4 version 1.9.0 - rc2: https://discuss.px4.io/t/default-s500-airframe-configuration-not-flyable-with-default/12394/4

closed 2022-11-22 by junwoo091400

MaEtUgR, 2022-12-06:
Ouuf, I think it's fair to close this because since 1.9 we fixed a lot of things that could have caused this.