PX4/PX4-Autopilot #13055: Using message_interval on any message stops sys_status from being transmitted

Crowdedlight, 2019-09-30:
**Describe the bug**
``sys_status`` do not get transmitted after using a ``message_interval`` to limit any message. I have tried limiting different messages, but no change, and ``sys_status`` is still not being transmitted. I have also tried to send ``message_interval`` for ``sys_status`` to try and force it to 1 or 5hz. But didn't have any impact. 

I checked with QGC and mavlink_inspector to verify if the message gets output. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Send a ``message_interval`` to limit some message
3. Open QGC and mavlink_inspector and sys_status isn't being transmitted, or with very low rate. (I sometimes got 1 maybe 2 messages over 30-40sec, but often didn't get any at all)

Rebooting the PX4 to output default messages makes the ``sys_status`` show up again, right until the next time you send ``message_interval``.

**Expected behavior**
``sys_status`` should not be "removed" when limiting some other message with ``message_interval``. 
Using ``message_interval`` for the ``sys_status`` message should correctly set the rate of the ``sys_status`` message.

**Drone (please complete the following information):**
- Hexacopter platform, PX4 1.9.2.

dagar, 2019-09-30:
Could you try this in current master?

Crowdedlight, 2019-10-01:
Tested on current master and tag: ``v1.10.0-beta1``. 
Same experience. After limiting any message I do not get ``sys_status``. I sometimes get a single ``sys_status`` 15s after the limit, but nothing more than that. 

We have also experienced that certain messages output by default can not be throttled by the ``message_interval`` command. Just gives ``MAV_RESULT_FAILED``. Seems like it is because they are not defined as a streamlist_item if I read the [source code correct](https://github.com/PX4/Firmware/blob/07d656e971a72d1202651dfd3b4642736fb078d7/src/modules/mavlink/mavlink_messages.cpp#L5050). 

We experienced it with the following msg ids: 
MAVLINK_MSG_ID_RC_CHANNELS_SCALED = 34      # kept failing  
MAVLINK_MSG_ID_RC_CHANNELS_RAW = 35         # kept failing  
MAVLINK_MSG_ID_RC_CHANNELS_OVERRIDE = 70    # kept failing  
MAVLINK_MSG_ID_BATTERY_STATUS = 147         # kept failing  
MAVLINK_MSG_ID_ESTIMATOR_STATUS = 230       # kept failing  

(I know this is probably another issue, but trying to give as much information as possible, if it helps)

Crowdedlight, 2019-10-03:
We have also experienced that using the ``message_interval`` mess with output of ``global_position_int``. When no ``message_interval`` has been sent so it output at default rates I receive ``global_position_int`` with 2hz, but as soon as ``message_interval`` is used, it drops down to inconsistent 0.3hz. 

This has only been tested on PX4 1.9.2.

dagar, 2019-10-03:
Could you try running `mavlink status` + `mavlink status streams` on the command line and capture the output before and after altering a message_interval?

Crowdedlight, 2019-10-04:
The console didn't let me copy, so I captured it as images. It clearly show that something happens with sys_status which makes it go down to a rate of 0.05hz. There also seems to be some odd values at ``system_time``, ``altitude``, ``extended_sys_state`` and ``estimator_status``. However they are all messages I try to limit to 1hz. I am gonna investigate if I send a wrong interval value for those messages. 

## Before limiting 
**Instance 0, is the one connected to my laptop when testing**
[image]
[image]

## After limiting
[image]
[image]

Crowdedlight, 2019-10-04:
It would appear that ``sys_status`` works as expected. The issue was me using the wrong unit for other messages. As such I assume it started to limits messages to stay below the overall limit of bytes/s, as it expected to send other messages with a million Hz. 

Apologies for the stupid error. Gonna close the issue here. Thanks for the help :)

closed 2019-10-04 by Crowdedlight