PX4/PX4-Autopilot #13061: RC override for VTOL

sfuhrer, 2019-10-01:
**Describe the bug**
Since https://github.com/PX4/Firmware/pull/12495 got merged 3 months ago, RC override is disabled for fixed-wing flight of a VTOL. It is though still enabled for hover flight. The current logic inside commander is such that if you move the sticks during the FW flight, and then don't move it back to where they were were originally, the RC override triggers a switch to manual (position) mode as soon as the back-transition happens. The most likely stick to cause that is the throttle stick, as it is often not spring-loaded.

I know of at least one case where the pilot moved the throttle stick during FW flight (probably to be prepared to take-over fast if necessary).

**To Reproduce**
In sitl VTOL:
- plan mission with VTOL_takeoff, some FW flight and then a transition to hover
- enable RC_override (COM_RC_OVERRIDE)
- move the throttle stick during the FW flight and keep it constant
- now, the RC override triggers as soon as the backtransition happens

**Expected behavior**
I would expect that only movements on the RC sticks during actual hover trigger the RC override.

MaEtUgR, 2019-10-06:
Shit, that's a corner case I never went into while testing since I did it with a spring-loaded joystick that was always in middle position when I back transitioned.

jkflying, 2020-08-05:
Potential fix: https://github.com/PX4/Firmware/pull/15495

MaEtUgR, 2020-11-19:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/15495

closed 2020-11-19 by MaEtUgR