PX4/PX4-Autopilot #13078: PX4FLOW is not detected since v1.10.0

Jeoker, 2019-10-02:
Hi all,

I am working with fmu-v5 and px4flow. It was working ok in firmware v1.9.0. However, when I update the firmware to v1.10.0, the px4flow is not detected anymore. I tried roll back to v1.9.0 and px4flow is detected again. I guess this is a bug in the new firmware?

dagar, 2019-10-02:
Hi Jeoker, 

A new parameter was added to start px4flow to save time needlessly probing for it every single bootup.
https://github.com/PX4/Firmware/blob/d537f3ec6ba2a58188869481f1dbcd36dc447a36/src/drivers/optical_flow/px4flow/parameters.c#L42

Jeoker, 2019-10-02:
Oh, thank you for pointing it out!

closed 2019-10-02 by Jeoker

hamishwillee, 2019-10-02:
@dagar Already documented in very first sentence of doc setup on master docs (v1.1.0 not out, so no specific docs for that yet): http://docs.px4.io/master/en/sensor/px4flow.html#pixhawk-setup