PX4/PX4-Autopilot #13083: CUAV V5+_Accel inconsistent

Tony3dr, 2019-10-02:
**Describe the bug**
We are getting this constant error message of Accel inconsistent error. We did all the calibrations, then try to arm again and again mark Accel inconsistent. We did a complete reset in the flight controller did the calibrations and the same issue. 

**Log Files and Screenshots**
[image]

Log: 
https://review.px4.io/plot_app?log=37067a71-b746-4f44-8013-8c4fe57b3bfa

Last working flight log: 
https://review.px4.io/plot_app?log=66332487-04b5-47f5-b1a3-a241e23a4dd7  

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- FC- CUAV V5+
- Frame- DJI f450
- GPS- NEO V2
[image]

**Additional context**
Today we tried the Master branch but I had already tried it in Stable & a PR and still the same error, I also changed the GPS from the working unit Nano CUAV but it remains the same.

LorenzMeier, 2019-10-03:
@MaEtUgR This is a real issue that we should solve with priority.

dagar, 2019-10-04:
One minor thing to consider here is if we can now afford to log all sensor data by default at a low rate. Currently these issues are often difficult or impossible to debug in default logs.

Zainpeng, 2019-10-08:
@Tony3dr Is this a mistake in the constant use of flight control? And then it doesn't disappear?

Tony3dr, 2019-10-14:
> @Tony3dr Is this a mistake in the constant use of flight control? And then it doesn't disappear?
@Zainpeng I would not say constant use since the error message does not let us fly. Since the issue started showing we are not able to use that flight controller.

dagar, 2020-04-15:
This appears to be a real issue with the icm20689 on the CUAV v5+ (and nano).

dagar, 2020-06-06:
Fixed in https://github.com/PX4/Firmware/pull/15004.

closed 2020-06-06 by dagar