PX4/PX4-Autopilot #13092: [VTOL] Offboard attitude control not working

ThomasRigi, 2019-10-04:
**Describe the bug**
Based on [discussion 13159](https://discuss.px4.io/t/vtol-hil-with-offboard-attitude-setpoints-not-working/13159).

VTOLs don't follow the offboard attitude setpoints, instead they keep the last attitude setpoint from before switching to offboard mode.

For more details (and other issues encountered), see the above-mentioned post on discuss.px4.io.

**To Reproduce**
Steps to reproduce the behavior:
Run the [MAVSDK offboard example](https://mavsdk.mavlink.io/develop/en/examples/offboard_velocity.html).

I have tested it on SITL and HIL on the standard VTOL model (with a slightly modified version of the MAVSDK example), in both simulations there are issues. Haven't dared to try it on a real drone, pretty sure it also doesn't work.

**Expected behavior**
Drone should follow the attitude setpoints that are passed via MAVLink.

**Log Files and Screenshots**
See [post](https://discuss.px4.io/t/vtol-hil-with-offboard-attitude-setpoints-not-working/13159) on Discuss.

**Drone (please complete the following information):**
- Standard VTOL (Gazebo)

**Additional context**

Main findings so far:
- The problem can be solved for SITL by settting the correct ORB ID (`mc_virtual_attitude_setpoint`) for `_att_sp_pub` in `mavlink_receiver.h` (need to set it so that it still works for non-VTOLs)
- The SITL solution doesn't work for HIL (tested with Pixhawk 4 as hardware). (not tested on real vehicles)
- Additionally, the correct frametype has to be included [here](https://github.com/PX4/Firmware/blob/f885d2274a563278b5ca501e4c19365ad0cd790d/src/modules/mavlink/mavlink_receiver.cpp#L1387) for the thrust setpoint.

hamishwillee, 2019-10-06:
Is this on master or on a stable build? Note that there was recently quite a lot of work done on offboard mode for non-copter targets by @Jaeyoung-Lim (Jae - FYI).

ThomasRigi, 2019-10-07:
@hamishwillee  On master and on v1.9.2

Jaeyoung-Lim, 2019-10-07:
@ThomasRigi Could you verify what typemask you are sending and if it is multicopter mode you are observing this behavior?

ThomasRigi, 2019-10-07:
@Jaeyoung-Lim The typemask is 0b00000111 (Ignore body roll, pitch and yaw rates), sent using MAVSDK function `Offboard::set_attitude(Offboard::Attitude attitude).` And yes, it is Multicopter mode

Jaeyoung-Lim, 2019-10-08:
@ThomasRigi It seems like this is specific to attitude setpoints. I can fly with body rate setpoints, which uses the same mavlink message (SET_ATTITUDE_TARGET)[https://mavlink.io/en/messages/common.html#SET_ATTITUDE_TARGET]. [here](https://review.px4.io/plot_app?log=630abfab-a6ca-4862-8998-7c1c0fafc3cc) is the log.

ThomasRigi, 2019-10-08:
@Jaeyoung-Lim Thanks for sharing the log! It makes sense that the rate control works, as in that case you don't have to pass through the virtual attitude setpoints in `vtol_att_control_main.cpp`. 

I will shortly create a PR in which SITL works (but HIL doesn't), but I know that @dagar doesn't like creating extra ORB publications.

I failed to figure out where this difference SITL <--> HIL comes from. It would be great if someone finds it (or dares to test the new solution on a real drone :grimacing:)

ThomasRigi, 2019-10-09:
#13141 is now tackling this problem. Many thanks to @irsdkv !

I'm crossing my fingers that HIL is going to work on his branch :crossed_fingers:

Jaeyoung-Lim, 2020-01-07:
This has been fixed in #13141

closed 2020-01-07 by Jaeyoung-Lim