PX4/PX4-Autopilot #13094: [Fixedwing]mission_landing_required should not relative to RTL type

jinchengde, 2019-10-05:
**Describe the bug**
fixed wing update flight plan, QGC error "Mission rejected: landing pattern required"

**To Reproduce**
Steps to reproduce the behavior:
1. start sitl gazebo plane
2. set the flight plan without land waypoint
3. upload
4. error "Mission rejected: landing pattern required"

**Additional context**
in check_mission_valid()
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1644
check the navigator mission_landing_required()
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator.h#L274
just return the RTL type whether RLT type is equal RTL_LAND

I am not clear whether RTL type should be combind the mission land required

Antiheavy, 2019-10-05:
Recently the fixed wing default parameters were updated to default RTL_TYPE to 1 which is to Return via a mission landing pattern.  This is the best RTL_TYPE for the vast majority of fixed wing systems.   This safety check rejects any mission that does not have a landing pattern, because how else will you get back and land when you lose Comms or go low battery?

jinchengde, 2019-10-05:
but in some use case like land survey, we wish the last action is loiter at home and open the parachute, and another question is when we upload the flight plan, qgc just prompt no Land pattern but if we set the last waypoint as land, no effective because there is no land start. no person will find the issue is RTL type before you check the source code.

Antiheavy, 2019-10-06:
>  we wish the last action is loiter at home and open the parachute,

In this case you will want to change the RTL_TYPE parameter to something other than the fixed wing default of 1, e.g. probably set it to 0.  The fixed wing defaults are set here:  https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.fw_defaults#L33

> qgc just prompt no Land pattern but if we set the last waypoint as land, no effective because there is no land start. no person will find the issue is RTL type before you check the source code.

The text message sent from the autopilot to QGC used to say "land start required", but we changed it to "landing pattern required" because the general user had no idea what a "land start" was, but they generally do know what a "landing pattern" is because it is one of the mission planning buttons in Qgroundcontrol. 

You raise a good point here though about potential confusion for people just using a Land waypoint vs using a Landing Pattern.  The Mission Feasibility module checks for a "mission landing pattern" as documented in the Return Mode behavior docs here https://docs.px4.io/master/en/flight_modes/return.html 
[image]