PX4/PX4-Autopilot #13107: Fixedwing mission clear noise

julianoes, 2019-10-07:
**Describe the bug**
When the mission is cleared from a fixedwing in CI there is a lot of wrong noise about a missing landing pattern:
```
[ INFO] [1570433448.966859511, 93.460000000]: FCU: Landing detected
[ INFO] [1570433448.979790857, 93.472000000]: WP: reached #4
[ INFO] [1570433448.993149250, 93.484000000]: WP: reached #4
[ INFO] [1570433449.018671774, 93.512000000]: FCU: Mission finished, landed.
[ INFO] [1570433449.075340507, 93.568000000]: FCU: DISARMED by Auto disarm initiated
[ INFO] [1570433449.118738819, 93.612000000]: FCU: Executing Mission
[ INFO] [1570433449.167259426, 93.660000000]: FCU: Takeoff on runway
INFO  [logger] closed logfile, bytes written: 5071752
WARN  [navigator] Mission rejected: landing pattern required.
ERROR [navigator] mission check failed
[ INFO] [1570433450.269600297, 94.748000000]: WP: mission cleared
[ERROR] [1570433450.272883659, 94.752000000]: FCU: Mission rejected: landing pattern required.
[ INFO] [1570433450.331607708, 94.804000000]: FCU: Mission finished, landed.
```

**To Reproduce**
Run `test/rostest_px4_run.sh mavros_posix_test_mission.test mission:=FW_mission_1 vehicle:=plane`

Antiheavy, 2019-10-07:
Not only in CI, but in reality.  I expect this same text is happening on VTOL CI too since VTOL default RTO_TYPE was recently updated to be "1"?  
I tell people that the "mission rejected, landing pattern required" is a "good warning" that is helping to remind them they need to add a landing pattern to their mission (required by RTL_TYPE=1).  Of course, the text message interface is an annoying way to convey that status information.  This is all tied up with a couple loosely related issues having to do with fixed wing RTL_TYPE and safety:
1) Mission Clear is unsafe if you are in flight and RTL_TYPE=1, would it be possible to prevent Mission Clear if in-flight and only allow mission replacement?
2) There is a fundamental flaw in mission feasibility checker in that it runs AFTER the old mission has been replaced or cleared and you lose your safe RTL behavior as discussed here https://github.com/PX4/Firmware/issues/12473

hamishwillee, 2019-10-08:
@Antiheavy As the docs are written now the landing pattern is intended to be optional and the system will fall back to flying to home and circling (by default) if one is not present. Under that mode clearing missions is fine, though having a warning would be nice. 

When did this change?

hamishwillee, 2019-10-08:
PS End users are finding this very irritating because it isn't obvious why their missions are now being rejected and how they can fix this. Someone was complaining about rally points too, but they may be confused.

Antiheavy, 2019-10-08:
> PS End users are finding this very irritating because it isn't obvious why their missions are now being rejected and how they can fix this.

Yes, this is in our top 3 list of user complaints.

> As the docs are written now the landing pattern is intended to be optional and the system will fall back to flying to home and circling (by default) if one is not present. Under that mode clearing missions is fine, though having a warning would be nice.

The docs correctly describe the behavior.  "Fine" may depend on your perspective though.  If you want the system to auto-land at the location you planned when the aircraft took off (I'd say this is the expectation of pretty much all users), then clearing the mission isn't fine.  Users aren't typically knowledgeable enough to realize clearing the mission will blow away their safe landing location in case of Return Mode.  Fixing this properly requires addressing https://github.com/PX4/Firmware/issues/12473 and probably also preventing the mission clear command in flight.

hamishwillee, 2019-10-09:
@Antiheavy Thanks for clarifying. Lots of edge cases.

So when did this check for upload go in? - ie can you answer https://github.com/PX4/Firmware/issues/13107#issuecomment-539257737

The reason I ask is that even the RTL params do not mention this behaviour. They indicate the fallback to flying back to home for this case. That might be what happens still, but we should mention the mission upload requirement in docs.

Antiheavy, 2020-01-07:
> So when did this check for upload go in? - ie can you answer [#13107 (comment)](https://github.com/PX4/Firmware/issues/13107#issuecomment-539257737)

@hamishwillee sorry for the delayed reply.  The mission feasibility check went in with the RTL_TYPE = 1 (use mission landing pattern) back in 1.8.0 or maybe earlier (work had been going on with it back in the 1.6/1.7 days).   Also, this new PR is somewhat related to this topic of RTL_TYPE=1 requiring a mission landing: https://github.com/PX4/Firmware/pull/13873

hamishwillee, 2020-01-07:
Thanks @Antiheavy . Things are even more messy now with the addition of Rally points. I'm not sure this check even makes sense since you will use a rally point in preference to a mission landing if it is closer.

Antiheavy, 2020-01-08:
> Things are even more messy now with the addition of Rally points. I'm not sure this check even makes sense since you will use a rally point in preference to a mission landing if it is closer.

I agree its more messy now with Rally points.  However, landing at Rally Points and/or the Home point are not appropriate for the vast majority of fixed wing use cases - so this mission feasibility check is still required until a better solution can be developed.

hamishwillee, 2020-01-08:
Fair enough. I think we need "Fixed wing landing points" or "Fixed wing rally points" that are independent of missions.

Antiheavy, 2020-01-08:
> I think we need "Fixed wing landing points" or "Fixed wing rally points" that are independent of missions.

This would be a great way to solve multiple problems!  Would be a major undertaking though.