PX4/PX4-Autopilot #13116: Vision position estimation fails depending on initial yaw (in Gazebo-SITL)

michael-p, 2019-10-07:
**Describe the bug**
Changing the initial rotation (yaw) of the copter `iris_vision` in Gazebo SITL causes the vision position estimation to fail. With different initial yaw like the default in `iris_vision.world` it works, albeit very shaky.

**To Reproduce**
Modify `iris_vision.world` by rotating the model:
```
[...]
<include>
      <uri>model://iris_vision</uri>
      <!-- Original: <pose>1.01 0.98 0.83 0 0 1.14</pose> -->
      <pose>1.01 0.98 0.83 0 0 -1.5</pose>
</include>
[...]
```
Run `make px4_sitl gazebo_iris_vision`, switch to position mode and take off, hover a few meters above ground. After a few seconds the copter flys away full speed, and then the failsafe kicks in and switches it to Altitude mode.
Console output:
```
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 14, OBS buf: 9)
INFO  [ecl/EKF] 1432000: EKF commencing external vision position fusion
INFO  [ecl/EKF] 1432000: EKF commencing external vision yaw fusion
INFO  [commander] Takeoff detected
WARN  [commander] Failsafe enabled: no local position
```

**Expected behavior**
Not sure, honestly :) Even with the default initial yaw of the model the copter is very unstable so maybe the noise/drift parameters of the Gazebo vision plugin are just too extreme? I don't know. However, I would at least expect the behavior to be independent of the initial yaw of the copter, that is, it should either always work or always fail (after some time maybe).

TSC21, 2019-10-07:
@michael-p what version of the Firmware are you using?

michael-p, 2019-10-08:
I tried this with the current master, but was also able to reproduce it with an older version (at commit 190c817a9e).
Sadly the latest stable release 1.9.2 does not compile so I cannot try that.

julianoes, 2019-10-08:
> Sadly the latest stable release 1.9.2 does not compile so I cannot try that.

Why? You likely just need to remove `-Werror`.

FYI @kamilritz

michael-p, 2019-10-08:
> Why? You likely just need to remove -Werror.

Yes I believe this was fixed in https://github.com/PX4/Firmware/pull/12108, that's why I did not open an issue for it.

I just confirmed that in v1.9.2 the problem exists as well, although the console output is slightly different: (the ERROR line is different)
```
INFO  [commander] Takeoff detected
ERROR [commander] Critical navigation failure! Check sensor calibration
WARN  [commander] Failsafe enabled: no local position
```

mhkabir, 2020-01-06:
I think I just ran into this as well. Pending further testing. @tzai can you try to reproduce using the iris_vision model in vanilla SITL (no ROS)?

kamilritz, 2020-01-27:
I think this should solve the issue: https://github.com/PX4/sitl_gazebo/pull/399

closed 2020-04-26 by kamilritz