PX4/PX4-Autopilot #13117: Build error GSTREAMER

baumanta, 2019-10-08:
**Describe the bug**
When trying to build the target `make px4_sitl_default gazebo` from current master, I get the following error:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "LiftDragPlugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/tanja/src/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/tanja/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/tanja/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[225/690] Linking CXX static library src/modules/uORB/libmodules__uORB.a
FAILED: cd /home/tanja/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/tanja/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/tanja/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[225/690] Linking CXX static library src/modules/uORB/libmodules__uORB.a
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

I bisected the branch and the problem starts at:

commit 91badceb274a0e5bc29061dffa6c124ae8285022
Author: TSC21 <n.marques21@hotmail.com>
Date:   Sun Oct 6 11:01:45 2019 +0100

    Update submodule sitl_gazebo to latest Sun Oct 06 11:02:54 GMT 2019

I am running this under Ubumtu 16.04 with GStreamer Core Library version 1.8.3

TSC21, 2019-10-08:
You need to install gstreamer1.0.

baumanta, 2019-10-08:
@TSC21 do you have any idea what could have caused that and how to fix it? I tried upgrading gstreamer on my machine but somehow did not manage. I don't know if 1.8.3 is the default supported for Ubuntu 16.04.

TSC21, 2019-10-08:
Gstreamer 1.0 is now a sitl_gazebo dependency. The version you need is 1.0, and not the latest: https://github.com/PX4/sitl_gazebo#gstreamer-support.

baumanta, 2019-10-08:
The instructions in the guide did not work for me. Executing `sudo apt-get install gstreamer1.0-* libgstreamer1.0-*` gave the following error:

The following packages have unmet dependencies:
 gstreamer1.0-hybris:i386 : Depends: libmedia1:i386 but it is not going to be installed

But installing: `sudo apt-get install libgstreamer-plugins-base1.0-dev`
seems to fix the problem. I can now build it again

closed 2019-10-08 by baumanta

TSC21, 2019-10-08:
> The instructions in the guide did not work for me. Executing
> `sudo apt-get install gstreamer1.0-* libgstreamer1.0-*` gave the following error:
>
> The following packages have unmet dependencies: [...]

Yeah looks like those need to be updated.

chricloud9, 2020-04-12:
I get this same issue, but when I ran sudo apt-get install libgstreamer-plugins-base1.0-dev and build the target again, I get Module Not Found Errors

ModuleNotFoundError: No module named 'jinja2'

I think it may have something to do with the build trying to use python 3? I have python 3 on my machine, but 2.7 is the default. Here's why i think it may be python 3 related.

[6/723] Generating serial_params.c
FAILED: generated_params/serial_params.c 
cd /home/user/src/Firmware/build/px4_sitl_default/src/lib/parameters && /usr/bin/cmake -E make_directory /home/user/src/Firmware/build/px4_sitl_default/generated_params && /usr/bin/python3

hamishwillee, 2020-04-14:
@chricloud9 Please don't necrobump dead issues. It is irritating. Your problem is that you haven't installed the PX4 development environment properly - see http://dev.px4.io/master/en/setup/dev_env.html

swimmingseeds, 2023-03-14:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/px4-sitl-gazebo-syntax-error-error-in-cmakelists-txt-file/31212/7