PX4/PX4-Autopilot #13118: Unable to biuld px4 v 1.9.0 by cross-compilation on Ubuntu16.04

MartinAtCoventry, 2019-10-08:
**Describe the bug**
I am developing a PX4 flight controller which based on BeagleBoneBlue platform. I have set a cross compilation environment according to [](https://docs.px4.io/v1.9.0/en/flight_controller/beaglebone_blue.html)
But there still are some compilation errors when I execute a cross-compilation task on my Ubuntu(16.04LTS) development host.

**To Reproduce**
1. copy robotcontrol.h and rc/* from my beagleboneblue board into /home/yangyang/px4/robotcontrolLib/include(on my Ubuntu host).

2. copy librobotcontrol.* from beagleboneblue board into /home/yangyang/px4/robotcontrolLib/lib.

3. export LIBROBOTCONTROL_INSTALL_DIR=/home/yangyang/px4/robotcontrolLib

4. export PATH=$PATH:/opt/compilers/gcc-arm-linux-gnueabihf/bin
    export CrossCompiler=/opt/compilers/gcc-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-
    Tip: the version of cross-compiler is gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf

5. clone the px4 firmware (1.9.0-rc2) to home/yangyang/px4/Firmware

6. cd px4/Firmwre, and run "make beaglebone_blue_cross".

7. When compiling [6/775], see the following errors.

**Expected behavior**
Go on with the cross compilation and finish it so that px4 can be used on the beagleboneblue. 

**Log Files and Screenshots**
make beaglebone_blue_cross
-- PX4 config file: /home/yangyang/px4/Firmware/boards/beaglebone/blue/cross.cmake
-- PX4 config: beaglebone_blue_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.1
-- The C compiler identification is GNU 7.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /home/yangyang/py2env/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- RTPS send: home_position;wind_estimate;radio_status;satellite_info;adc_report;iridiumsbd_status;vtol_vehicle_status;airspeed;sensor_selection;cpuload;sensor_baro;battery_status;estimator_status;distance_sensor;vehicle_odometry;vehicle_attitude;collision_constraints;sensor_combined
-- RTPS receive: trajectory_waypoint;position_setpoint;camera_capture;collision_report;obstacle_distance;debug_array;debug_key_value;debug_vect;position_setpoint_triplet;debug_value;camera_trigger;vehicle_trajectory_waypoint;optical_flow
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yangyang/px4/Firmware/build/beaglebone_blue_cross
[6/755] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
FAILED: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../src/platforms/common -I../../platforms/posix/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -funsafe-math-optimizations -fno-strict-aliasing -fno-math-errno -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -I/home/yangyang/px4/robotcontrolLib/include -std=gnu++11 -MMD -MT src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o -MF src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o.d -o src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o -c ../../src/lib/DriverFramework/framework/src/DriverFramework.cpp
../../src/lib/DriverFramework/framework/src/DriverFramework.cpp: In function ‘void show_sched_settings()’:
../../src/lib/DriverFramework/framework/src/DriverFramework.cpp:311:21: error: variable ‘show_sched_settings()::sched_param param’ has initializer but incomplete type
  struct sched_param param {};
                     ^~~~~
compilation terminated due to -Wfatal-errors.
[6/755] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1

Thanks in advance

julianoes, 2019-10-08:
Have you tried to add `#include <sched.h>`?

See: http://man7.org/linux/man-pages/man2/sched_setparam.2.html

MartinAtCoventry, 2019-10-08:
**Describe the bug**
@julianoes , thank you a lot for my issue.
Now, the problem "DriverFramework.cpp:311:21: error: variable ‘show_sched_settings()::sched_param param’ has initializer but incomplete type
  struct sched_param param {};" has been solved after adding "#include <sched.h>" into the file DriverFramework.cpp.
But there still are other errors while going on to cross-compiling the PX4 firmware project.

**To Reproduce**
In order to solve "error: ‘pthread_self’ was not declared in this scope", I added the compilation flag "-pthread" into Firmware/cmake/px4_add_common_flags.cmake. But the error is still here. I don't know have any way to solve it.

**Log Files and Screenshots**
make beaglebone_blue_cross 
-- PX4 config file: /home/yangyang/px4/Firmware/boards/beaglebone/blue/cross.cmake
-- PX4 config: beaglebone_blue_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.1
-- The C compiler identification is GNU 7.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/yangyang/py2env/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- RTPS send: home_position;wind_estimate;radio_status;satellite_info;adc_report;iridiumsbd_status;vtol_vehicle_status;airspeed;sensor_selection;cpuload;sensor_baro;battery_status;estimator_status;distance_sensor;vehicle_odometry;vehicle_attitude;collision_constraints;sensor_combined
-- RTPS receive: trajectory_waypoint;position_setpoint;camera_capture;collision_report;obstacle_distance;debug_array;debug_key_value;debug_vect;position_setpoint_triplet;debug_value;camera_trigger;vehicle_trajectory_waypoint;optical_flow
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yangyang/px4/Firmware/build/beaglebone_blue_cross
[6/755] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
FAILED: /usr/bin/ccache /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../src/platforms/common -I../../platforms/posix/include -O2 -g -DNDEBUG -fPIC   -g -pthread -fdata-sections -ffunction-sections -fomit-frame-pointer -funsafe-math-optimizations -fno-strict-aliasing -fno-math-errno -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -I/home/yangyang/px4/robotcontrolLib/include -std=gnu++11 -MMD -MT src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o -MF src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o.d -o src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o -c ../../src/lib/DriverFramework/framework/src/DriverFramework.cpp
../../src/lib/DriverFramework/framework/src/DriverFramework.cpp: In function ‘void show_sched_settings()’:
../../src/lib/DriverFramework/framework/src/DriverFramework.cpp:314:34: error: ‘pthread_self’ was not declared in this scope
  int ret = pthread_getschedparam(pthread_self(), &policy, &param);
                                  ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
[6/755] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1

julianoes, 2019-10-09:
Have you tried adding `#include <pthread.h>` in the files where `pthread...` is not declared?

MartinAtCoventry, 2019-10-09:
@julianoes , the above errors have been solved through adding `#include <pthread>` into `Firmware/src/lib/DriverFramework/framework/src/DriverFramework.cpp`. Also, the flag`-pthread` can be removed from `Firmware/cmake/px4_add_common_flags.cmake`, and the
cross-compilation is not impacted as well.

However, I have only one error at the last step.
**Describe the bug**
The cross-compiler cannot find the robotcontrol library.

**To Reproduce**
1. copy `robotcontrol.h` and `rc/*` from my beagleboneblue board into `/home/yangyang/px4/robotcontrolLib/include`(on my Ubuntu host).
2. copy `librobotcontrol.*` from beagleboneblue board into `/home/yangyang/px4/robotcontrolLib/lib`
3.`export LIBROBOTCONTROL_INSTALL_DIR=/home/yangyang/px4/robotcontrolLib`
You can see https://docs.px4.io/v1.9.0/en/flight_controller/beaglebone_blue.html

**Expected behavior**
Find a way to correctly set the library files of the robotcontrol library into a right place. Let the cross-compilation finish.

**Log Files and Screenshots**
```
[755/755] Linking CXX executable bin/px4
FAILED: : && /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -O2 -g -DNDEBUG  -Wl,--export-dynamic platforms/posix/CMakeFiles/px4.dir/src/main.cpp.o platforms/posix/CMakeFiles/px4.dir/apps.cpp.o  -o bin/px4 -L/home/yangyang/px4/Firmware/build/beaglebone_blue_cross/external/Install/lib -rdynamic src/lib/cdev/test/liblib__cdev__test__cdev_test.a src/lib/controllib/controllib_test/liblib__controllib__controllib_test.a src/lib/rc/rc_tests/liblib__rc__rc_tests.a src/modules/uORB/libmodules__uORB.a src/modules/uORB/uORB_tests/libmodules__uORB__uORB_tests.a boards/beaglebone/blue/bbblue_adc/libdrivers__bbblue_adc.a src/drivers/batt_smbus/libdrivers__batt_smbus.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/differential_pressure/ets/libdrivers__ets_airspeed.a src/drivers/differential_pressure/ms4525/libdrivers__ms4525_airspeed.a src/drivers/differential_pressure/ms5525/libdrivers__ms5525_airspeed.a src/drivers/differential_pressure/sdp3x/libdrivers__sdp3x_airspeed.a src/drivers/distance_sensor/cm8jl65/libdrivers__cm8jl65.a src/drivers/distance_sensor/leddar_one/libdrivers__leddar_one.a src/drivers/distance_sensor/ll40ls/libdrivers__ll40ls.a src/drivers/distance_sensor/mb12xx/libdrivers__mb12xx.a src/drivers/distance_sensor/pga460/libdrivers__pga460.a src/drivers/distance_sensor/sf0x/libdrivers__sf0x.a src/drivers/distance_sensor/sf0x/sf0x_tests/libdrivers__sf0x__sf0x_tests.a src/drivers/distance_sensor/sf1xx/libdrivers__sf1xx.a src/drivers/distance_sensor/srf02/libdrivers__srf02.a src/drivers/distance_sensor/teraranger/libdrivers__teraranger.a src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a src/drivers/distance_sensor/ulanding/libdrivers__ulanding.a src/drivers/distance_sensor/vl53lxx/libdrivers__vl53lxx.a src/drivers/gps/libdrivers__gps.a src/drivers/pwm_out_sim/libdrivers__pwm_out_sim.a src/drivers/linux_pwm_out/libdrivers__linux_pwm_out.a src/drivers/linux_sbus/libdrivers__linux_sbus.a src/modules/attitude_estimator_q/libmodules__attitude_estimator_q.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/commander/commander_tests/libmodules__commander__commander_tests.a src/modules/dataman/libmodules__dataman.a src/modules/ekf2/libmodules__ekf2.a src/modules/events/libmodules__events.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/gnd_att_control/libmodules__gnd_att_control.a src/modules/gnd_pos_control/libmodules__gnd_pos_control.a src/modules/land_detector/libmodules__land_detector.a src/modules/landing_target_estimator/libmodules__landing_target_estimator.a src/modules/load_mon/libmodules__load_mon.a src/modules/local_position_estimator/libmodules__local_position_estimator.a src/modules/logger/libmodules__logger.a src/modules/mavlink/libmodules__mavlink.a src/modules/mavlink/mavlink_tests/libmodules__mavlink__mavlink_tests.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/micrortps_bridge/micrortps_client/libmodules__micrortps_bridge__micrortps_client.a src/modules/navigator/libmodules__navigator.a src/modules/sensors/libmodules__sensors.a src/modules/sih/libmodules__sih.a src/modules/vmount/libdrivers__vmount.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/wind_estimator/libmodules__wind_estimator.a src/systemcmds/esc_calib/libsystemcmds__esc_calib.a src/systemcmds/led_control/libsystemcmds__led_control.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/motor_ramp/libsystemcmds__motor_ramp.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/sd_bench/libsystemcmds__sd_bench.a src/systemcmds/tests/libsystemcmds__tests.a src/systemcmds/tests/hrt_test/libsystemcmds__tests__hrt_test.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/topic_listener/libsystemcmds__topic_listener.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/systemcmds/ver/libsystemcmds__ver.a src/examples/bottle_drop/libmodules__bottle_drop.a src/examples/fixedwing_control/libexamples__fixedwing_control.a src/examples/hello/libexamples__hello.a src/examples/position_estimator_inav/libmodules__position_estimator_inav.a src/examples/px4_mav [... cut ...] pper && cd /home/yangyang/px4/Firmware/build/beaglebone_blue_cross/bin && /usr/bin/cmake -E create_symlink px4 px4-df_bmp280_wrapper
/opt/compilers/gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/7.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lrobotcontrol
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1
```

julianoes, 2019-10-09:
It can't find `librobotcontrol.a` or `librobotcontrol.so`. You can try to add the path of where the file is using https://cmake.org/cmake/help/v3.0/command/link_directories.html in cmake.

MartinAtCoventry, 2019-10-09:
I have set the `LIBROBOTCONTROL_INSTALL_DIR` environment variable in my Ubuntu host according to https://docs.px4.io/v1.9.0/en/flight_controller/beaglebone_blue.html. The detail content look like the follow screenshot:
[image]
So, I thought Cmake should know where it find the robotcontrol library.

@julianoes , where and which file do I edit to add `link_directories(directory1 directory2 ...)` command into?

TSC21, 2019-10-09:
@MartinAtCoventry did you set `CLFAGS`?

julianoes, 2019-10-09:
Can you just add `link_directories(/home/yangyang/px4/robotcontrolLib/lib)` to the root `CMakeLists.txt`

MartinAtCoventry, 2019-10-09:
@TSC21 , how can I set the CFLAG?

MartinAtCoventry, 2019-10-10:
**Describe the bug**
@julianoes , thank you for help me. Now, Cmake can find the path of the robotcontrol dynamic library. I've saw the line info:`-L/home/yangyang/px4/robotcontrolLib/lib` in compilation information. However, I still meet another problem:
`Firmware/build/beaglebone_blue_cross/../../src/lib/DriverFramework/framework/src/I2CDevObj.cpp:176: undefined reference to 'rc_i2c_read_data'`

I've found the function `rc_i2c_read_data()` in the file `Firmware/src/lib/DriverFramework/framework/src/I2CDevObj.cpp` is referenced from the header `robotcontrol.h`(http://strawsondesign.com/docs/librobotcontrol/i2c_8h_source.html).

So, I think this means the header files of the robotcontrol library are not found by Cmake yet. 

**Expected behavior**
I've putted the robotcontrol header files at the path `/home/yangyang/px4/robotcontrolLib/include` in my Ubuntu host.
How can I make Cmake finding the path these header files at?

**Log Files and Screenshots**
```
src/lib/DriverFramework/framework/src/libdf_driver_framework.a(I2CDevObj.cpp.o): In function 'DriverFramework::I2CDevObj::_readReg(unsigned char, unsigned char*, unsigned int)':
/home/yangyang/px4/Firmware/build/beaglebone_blue_cross/../../src/lib/DriverFramework/framework/src/I2CDevObj.cpp:176: undefined reference to 'rc_i2c_read_data'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
```

julianoes, 2019-10-11:
It looks like you're missing an update to DriverFramework. Can you check if you can checkout master of DriverFramework:
https://github.com/PX4/DriverFramework/commit/c7d2fe2d4f5954b9042af4edf7e5aeb21dbdd129

You can do this by:
```
cd src/lib/DriverFramework
git checkout master
git pull
cd -
```

And then, when you build, you need to say that you know what you're doing regarding submodules, otherwise the script will reset it back.

MartinAtCoventry, 2019-10-14:
@julianoes , thank you for helping me to solve my cross-compilation errors. Now, I have been able to successfully cross-compile a full set of PX4 flight stack for my BeagleboneBlue board.

MartinAtCoventry, 2019-10-14:
@julianoes , according to the way(https://github.com/PX4/DriverFramework/commit/c7d2fe2d4f5954b9042af4edf7e5aeb21dbdd129) you give, I have successfully cross-compiled the full project codes for my Beaglebone Blue (`v1.9.0-rc2`).
But there still were some cross-compilation errors when I checkout to the branch `Master`. 

**Log Files and Screenshots**
```
make beaglebone_blue_cross 
-- PX4 version: v1.10.0-beta2-95-g04ea849
-- PX4 config file: /home/yangyang/Firmware/boards/beaglebone/blue/cross.cmake
-- PX4 config: beaglebone_blue_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.1
-- The C compiler identification is GNU 7.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/yangyang/py2env/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-63-gf005e0e
-- 
-- RTPS send: home_position, wind_estimate, radio_status, satellite_info, adc_report, iridiumsbd_status, vtol_vehicle_status, airspeed, sensor_selection, cpuload, sensor_baro, battery_status, estimator_status, distance_sensor, vehicle_odometry, vehicle_attitude, collision_constraints, sensor_combined, 
-- RTPS receive: trajectory_waypoint, position_setpoint, camera_capture, collision_report, obstacle_distance, debug_array, debug_key_value, debug_vect, position_setpoint_triplet, debug_value, camera_trigger, vehicle_trajectory_waypoint, optical_flow, vehicle_mocap_odometry, vehicle_visual_odometry
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yangyang/Firmware/build/beaglebone_blue_cross
[28/780] git submodule src/drivers/gps/devices
[34/780] git submodule src/lib/DriverFramework
[36/780] git submodule src/lib/ecl
[38/780] git submodule mavlink/include/mavlink/v2.0
[39/780] git submodule src/modules/micrortps_bridge/micro-CDR
[475/780] Building CXX object src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o
FAILED: /usr/bin/ccache /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DMODULE_NAME=\"gps\" -DPX4_MAIN=gps_app_main -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wno-cast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -I/home/yangyang/px4/robotcontrolLib/include -std=gnu++11 -MMD -MT src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o -MF src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o.d -o src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o -c ../../src/drivers/gps/gps.cpp
../../src/drivers/gps/gps.cpp: In member function ‘virtual void GPS::run()’:
../../src/drivers/gps/gps.cpp:697:11: error: this statement may fall through [-Werror=implicit-fallthrough=]
     _mode = GPS_DRIVER_MODE_UBX;
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1
```

julianoes, 2019-10-14:
If you're on Ubuntu 16.04, try to update `ccache` as here:
https://github.com/PX4/Firmware/blob/04ea84908e1c465b89604b25976069a187763d28/Tools/setup/ubuntu.sh#L110-L112

MartinAtCoventry, 2019-10-15:
@julianoes , It's so fantastic, the above error disappeared after installing `ccache_3.4.1-1_amd64.deb`.
But there is a new error when I git the whole project (the master branch) and cross-compiled it again.

**Describe the bug**
In the file `Firmware/src/platforms/px4_log.h`, its macro define `__px4_log_modulename` has an undeclared macro `MODULE_NAME`.

**Expected behavior**
1. How can I solve this error?
2. Why the master branch has more cross-compilation errors than the v1.9.0-rc2 branch ?

**Log Files and Screenshots**
```
make beaglebone_blue_cross
-- PX4 version: v1.10.0-beta2-118-g97e1edc
-- PX4 config file: /home/yangyang/Firmware/boards/beaglebone/blue/cross.cmake
-- PX4 config: beaglebone_blue_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.1
-- The C compiler identification is GNU 7.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/yangyang/py3env/bin/python (found version "3.5.2") 
-- Found PY_jinja2: /usr/lib/python3/dist-packages/jinja2  
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-63-gf005e0e
-- 
-- RTPS send: distance_sensor, iridiumsbd_status, vehicle_odometry, vehicle_attitude, home_position, sensor_baro, estimator_status, cpuload, vtol_vehicle_status, collision_constraints, sensor_combined, satellite_info, sensor_selection, radio_status, wind_estimate, adc_report, battery_status, airspeed, 
-- RTPS receive: position_setpoint_triplet, camera_capture, debug_value, obstacle_distance, debug_vect, camera_trigger, optical_flow, collision_report, trajectory_waypoint, debug_key_value, position_setpoint, debug_array, vehicle_trajectory_waypoint, vehicle_mocap_odometry, vehicle_visual_odometry
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yangyang/Firmware/build/beaglebone_blue_cross
[27/780] git submodule src/drivers/gps/devices
[34/780] git submodule src/lib/DriverFramework
[35/780] git submodule src/lib/ecl
[36/780] git submodule mavlink/include/mavlink/v2.0
[38/780] git submodule src/modules/micrortps_bridge/micro-CDR
[38/780] Building C object boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o
FAILED: /usr/bin/ccache /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc  -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -Wno-error -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wno-cast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fno-common -Wbad-function-cast -Wnested-externs -Wstrict-prototypes -I/home/yangyang/px4/robotcontrolLib/include -std=gnu11 -MMD -MT boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o -MF boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o.d -o boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o   -c ../../boards/beaglebone/blue/src/init.c
In file included from ../../boards/beaglebone/blue/src/init.c:40:0:
../../boards/beaglebone/blue/src/init.c: In function ‘rc_init’:
../../src/platforms/px4_log.h:232:29: error: ‘MODULE_NAME’ undeclared (first use in this function)
   px4_log_modulename(level, MODULE_NAME, fmt, ##__VA_ARGS__); \
                             ^
../../src/platforms/px4_log.h:388:29: note: in expansion of macro ‘__px4_log_modulename’
 #define PX4_INFO(FMT, ...)  __px4_log_modulename(_PX4_LOG_LEVEL_INFO, FMT, ##__VA_ARGS__)
                             ^~~~~~~~~~~~~~~~~~~~
../../boards/beaglebone/blue/src/init.c:56:2: note: in expansion of macro ‘PX4_INFO’
  PX4_INFO("Initializing librobotcontrol ...");
  ^~~~~~~~
compilation terminated due to -Wfatal-errors.ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1
```

julianoes, 2019-10-15:
Add something like this before the include:
```
#ifndef MODULE_NAME
#define MODULE_NAME "bbbinit"
#endif
```

Sorry, I guess the BeagleBone is not really something that is regularly tested or maintained by the PX4 community.

MartinAtCoventry, 2019-10-15:
@julianoes , this seem incorrect. When added `#define MODULE_NAME "bbbinit"` into the file `/Firmware/src/platforms/px4_log.h`, the cross-compilation threw the following errors:
```
make beaglebone_blue_cross 
-- PX4 version: v1.10.0-beta2-118-g97e1edc
-- PX4 config file: /home/yangyang/Firmware/boards/beaglebone/blue/cross.cmake
-- PX4 config: beaglebone_blue_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.1
-- The C compiler identification is GNU 7.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/yangyang/py3env/bin/python (found version "3.5.2") 
-- Found PY_jinja2: /usr/lib/python3/dist-packages/jinja2  
DF Drivers: mpu9250;bmp280
Adding DF driver: mpu9250
Adding DF driver: bmp280
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-63-gf005e0e
-- 
-- RTPS send: adc_report, vtol_vehicle_status, sensor_selection, collision_constraints, sensor_baro, estimator_status, battery_status, wind_estimate, iridiumsbd_status, satellite_info, distance_sensor, airspeed, sensor_combined, vehicle_odometry, home_position, vehicle_attitude, cpuload, radio_status, 
-- RTPS receive: debug_vect, position_setpoint_triplet, camera_capture, debug_array, trajectory_waypoint, debug_key_value, vehicle_trajectory_waypoint, camera_trigger, collision_report, debug_value, obstacle_distance, optical_flow, position_setpoint, vehicle_mocap_odometry, vehicle_visual_odometry
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yangyang/Firmware/build/beaglebone_blue_cross
[28/780] git submodule src/drivers/gps/devices
[33/780] git submodule src/lib/DriverFramework
[36/780] git submodule src/lib/ecl
[38/780] git submodule mavlink/include/mavlink/v2.0
[39/780] git submodule src/modules/micrortps_bridge/micro-CDR
[582/780] Generating RTPS topic bridge[779/780] Building CXX object platforms/posix/CMakeFiles/px4.dir/apps.cpp.o
FAILED: /usr/bin/ccache /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DPX4_SHELL_COMMAND_PREFIX=\"px4-\" -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -Dpx4_EXPORTS -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -Iplatforms/posix -O2 -g -DNDEBUG -fPIE   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wno-cast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -I/home/yangyang/px4/robotcontrolLib/include -std=gnu++11 -MMD -MT platforms/posix/CMakeFiles/px4.dir/apps.cpp.o -MF platforms/posix/CMakeFiles/px4.dir/apps.cpp.o.d -o platforms/posix/CMakeFiles/px4.dir/apps.cpp.o -c platforms/posix/apps.cpp
platforms/posix/apps.cpp:14:0: error: "MODULE_NAME" redefined [-Werror]
 #define MODULE_NAME "px4"
 
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[779/780] Building CXX object platforms/posix/CMakeFiles/px4.dir/src/px4/common/main.cpp.o
FAILED: /usr/bin/ccache /opt/compilers/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++   -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DPX4_SHELL_COMMAND_PREFIX=\"px4-\" -DRC_AUTOPILOT_EXT -D__CUSTOM_FILE_IO__ -D__DF_BBBLUE -D__DF_BBBLUE_USE_RC_BMP280_IMP -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__PX4_POSIX_BBBLUE -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -Dpx4_EXPORTS -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../ [... cut ...]
 #define MODULE_NAME "px4"
 
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'beaglebone_blue_cross' failed
make: *** [beaglebone_blue_cross] Error 1
```
So, I changed the code you give because other two files have been defined `MODULE_NAME` as `"px4"`. My code looks like that:
```
#ifndef MODULE_NAME
#define MODULE_NAME "px4"
#endif
```
The whole project can be cross-compiled successfully.

MartinAtCoventry, 2019-10-15:
@julianoes , thank you so much for solving my cross-compilation problems. Through I cross-compiled again about the branch master and v1.9.0-rc2 of PX4 for BeagleboneBlue board, all of these two branches of code have been cross-compiled successfully.

Also, I forget to give you the binary files about the robotcontrol library. You can download them from my github address: 
https://github.com/MartinAtCoventry/Hello-world/blob/master/robotcontrolLib.tar.gz

closed 2019-10-15 by MartinAtCoventry