PX4/PX4-Autopilot #13120: macOS SITL build broken

julianoes, 2019-10-08:
After a fresh brew and brew dependencies install, I can't build for Gazebo SITL anymore.

```
<fill in error> 
```

julianoes, 2019-10-08:
It looks like we need to wait for https://github.com/osrf/homebrew-simulation/issues/850.

TSC21, 2019-10-08:
Yep I was struggling with the same this weekend on sitl_gazebo: https://travis-ci.org/PX4/sitl_gazebo/jobs/594207021

julianoes, 2019-10-08:
brew on macOS is like being on a rolling release Linux :cry:, it doesn't make it easy.

TSC21, 2019-10-09:
Solution should come with https://github.com/osrf/homebrew-simulation/pull/849

julianoes, 2019-10-11:
Great, this is fixed. Now I can continue to debug gstreamer and glib dependency issues.

closed 2019-10-11 by julianoes

TSC21, 2019-10-11:
@julianoes I am already working on that here if you want to check: https://github.com/PX4/sitl_gazebo/pull/353 It reaches the build stage but still fails because if the problem above.