PX4/PX4-Autopilot #13123: SITL ROS broken since sitl-gazebo submodule update

baumanta, 2019-10-08:
Since commit
commit 91badceb274a0e5bc29061dffa6c124ae8285022
Author: TSC21 <n.marques21@hotmail.com>
Date:   Sun Oct 6 11:01:45 2019 +0100

    Update submodule sitl_gazebo to latest Sun Oct 06 11:02:54 GMT 2019

The PX4/avoidance cannot be run anymore. Gazebo does not start up without giving any error message.
Running under Ubuntu 16.04 with gazebo7

baumanta, 2019-10-08:
@TSC21 do you have any clue what could have gone wrong?

TSC21, 2019-10-08:
This was reported and being solved in https://github.com/PX4/Firmware/pull/13113. I am already following with @mrivi on this.

baumanta, 2019-10-08:
It does not seem to be that same bug, the one you refer to was caused by bumping the CI container. The one I observe is only present since the mentioned commit of the sitl_gazebo update.

TSC21, 2019-10-08:
@baumanta I don't see anything on sitl_gazebo that may cause a potential. Can you please describe what errors do you see? Have you tried to run the simulation with `gdb`?

TSC21, 2019-10-08:
Besides, all the other tests on CI using Gazebo run as they should. So this needs to be tracked to the exact usage of Gazebo for Avoidance.

mrivi, 2019-10-08:
the avoidance tests are the only tests that simulate a camera and that has numerous implications.
The problem I see is the following: a container update broke the CI tests since the vehicle isn't spawing anymore. The PR was merged in a rushed way not allowing all the people involed to look at it and the tests were disabled. A subsequent PR that updates the sitl gazebo submodules breaks the avoidance for 16.04 setups. Those CI tests are in place exactly to catch this problems!

TSC21, 2019-10-08:
So let's track them.

baumanta, 2019-10-09:
update: the bug seems to be present on 18.04 and 16.04. It seems to be a corner case in the model spawning. Models containing lidars as well as depth cameras do not work. Both elements individually seem to be fine

mrivi, 2019-10-09:
@dagar - please read my comment above

TSC21, 2019-10-09:
> update: the bug seems to be present on 18.04 and 16.04. It seems to be a corner case in the model spawning. Models containing lidars as well as depth cameras do not work. Both elements individually seem to be fine

As we spoke, I am bisecting this directly on the code blocks. This is a corner case on models that contain both lidar (and probably sonar models) together with depth camera models. The same problem does not happen when the model include just the lidar or just the depth camera. So that's why this only happens with the avoidance models. The changes on the gazebo lidar plugin may be causing this, but it's not confirmed. After the source is found, sitl_gazebo will be updated again.

mrpollo, 2019-10-15:
Was this issue resolved with #13151 ?

mrivi, 2019-10-15:
fixed by #13113

closed 2019-10-15 by mrivi