PX4/PX4-Autopilot #13133: Compile Error  [-Werror=implicit-fallthrough=]

huangwen0907, 2019-10-09:
**Configure List**
~~~
hw0907@wen:~/Firmware$ make px4_fmu-v2_default 
[1/1] Re-running CMake...
-- PX4 version: v1.10.0-beta2-48-g6066300
-- PX4 config file: /home/hw0907/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- cmake build type: MinSizeRel
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-63-gf005e0e
-- Building and including px4_io-v2_default
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hw0907/Firmware/build/px4_fmu-v2_default
[3/620] Performing build step for 'px4io_firmware'
[1/1] Re-running CMake...
-- PX4 version: v1.10.0-beta2-48-g6066300
-- PX4 config file: /home/hw0907/Firmware/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- cmake build type: MinSizeRel
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-63-gf005e0e
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hw0907/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware

~~~

**Compile Error Show**
~~~
../../../../../src/modules/px4iofirmware/mixer.cpp
../../../../../src/modules/px4iofirmware/mixer.cpp: In function 'int mixer_handle_text(const void*, size_t)':
../../../../../src/modules/px4iofirmware/mixer.cpp:602:21: error: this statement may fall through [-Werror=implicit-fallthrough=]
   mixer_text_length = 0;
   ~~~~~~~~~~~~~~~~~~^~~

../../src/drivers/px4fmu/fmu.cpp: In member function 'int PX4FMU::set_mode(PX4FMU::Mode)':
../../src/drivers/px4fmu/fmu.cpp:298:13: error: this statement may fall through [-Werror=implicit-fallthrough=]
   PX4_DEBUG("MODE_2PWM2CAP");
   ~~~~~~~~~~^~~

s/drivers__ms5611.dir/ms5611.cpp.obj -c ../../src/drivers/barometer/ms5611/ms5611.cpp
../../src/drivers/barometer/ms5611/ms5611.cpp: In function 'int ms5611_main(int, char**)':
../../src/drivers/barometer/ms5611/ms5611.cpp:1082:5: error: this statement may fall through [-Werror=implicit-fallthrough=]
     }
     ^
~~~

**BTW**
``The px4_fmu-v4_default has the similar problem``

julianoes, 2019-10-09:
Can you check if it works with a newer ccache version?
https://github.com/PX4/Firmware/blob/c1faea8d32eca628c45d86bb6197b65c48441c0e/Tools/setup/ubuntu.sh#L110-L112

huangwen0907, 2019-10-09:
@julianoes  I have done as you recommend below and it build sucessfully. Thank you
```
 echo "Installing Ubuntu 16.04 PX4-compatible ccache version" 
 wget -O /tmp/ccache_3.4.1-1_amd64.deb http://launchpadlibrarian.net/356662933/ccache_3.4.1-1_amd64.deb 
 sudo dpkg -i /tmp/ccache_3.4.1-1_amd64.deb 
```

closed 2019-10-09 by huangwen0907

Cody-Chen, 2020-02-29:
Well done! It work for me too.