PX4/PX4-Autopilot #13139: Flight task error messages are not very helpful

julianoes, 2019-10-09:
The flight task error printfs on activation failures are verbose but still cryptic:

```
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Previous flight task failed, switching to mode 1
WARN  [mc_pos_control] Auto activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
```

FYI @Stifael @MaEtUgR

MaEtUgR, 2019-10-21:
There was supposed to be a bunch of possible errors e.g. Orbit activation failed with error: no global position or similar but the errors you see now are the results of a stub for this logic. They are better than no message but I agree with you totally confusing.

julianoes, 2024-03-04:
These have mostly disappeared, I think.

closed 2024-03-04 by julianoes