PX4/PX4-Autopilot #13147: Holybro ESC problems (can't fly with master)

dakejahl, 2019-10-09:
**Describe the bug**
Cannot fly my holybro s500 kit with `master`. It works fine on `v1.9.2`. I've also tracked this on px4 discuss [here](https://discuss.px4.io/t/holybro-esc-issue-on-master/12618/3).

**To Reproduce**
Using this holybro s500 kit: https://shop.holybro.com/_p1153.html

**Expected behavior**
There should be no ESC issues between PX4 firmware versions. There is funny business happening on the actuator outputs during boot. 

Below are screenshots of a scope on motor output 1 during first 1 second of boot.

**Log Files and Screenshots**
`v1.9.2`
[image]

`master`
[image]

As you can see in the images above, `v1.9.2` has 5 pulses of 900us width spaced at 20ms before the 400hz 900us outputs actually kick on. I am not sure if this is related, but it was the only thing I found. 

**Additional context**
This has been broken for a while now, I'd really like to be able to fly `master`...

julianoes, 2019-10-10:
@bkueng have you seen this?

davids5, 2019-10-10:
I have requested hardware so that I can help with this. but I will not be able to work on it till after I return on the 16th.

MaEtUgR, 2019-10-14:
We should have an assembled set. Let me check.

dagar, 2019-10-14:
> ould mean ours is s

@MaEtUgR could you try through a range of typical configurations?
 - 400 Hz PWM
 - OneShot
 - DShot

dakejahl, 2019-10-14:
Considering no one else but me is having this issue, I wonder about the ESC firmware. This might be something we ask Holybro about. Also Holybro has multiple ESC models, and I am not sure which model these are, the website does not list it.

jkflying, 2019-10-23:
It would be helpful to get a high-rate log, set this by enabling the flag in SDLOG_PROFILE, get a log with 1.9.2, do the same on master, and then upload the logs to compare.

MaEtUgR, 2019-10-23:
I'm currently trying to reproduce. Attached one of these black heat shrink ESCs of an S500 to a pixhawk 4, chose quad x and tried force arming and running `pwm test`. First result: The signal is on IO by default but @dakejahl mentioned he has the problem on fmu so I should probably load the S500 airframe.

dakejahl, 2019-10-23:
>  @dakejahl mentioned he has the problem on fmu
Sorry I think I misunderstood the question on the dev-call, I am indeed connected via IO port for motor outputs

MaEtUgR, 2019-10-23:
Here's a picture of the setup just for sanity check by @dakejahl :
[image]
[image]

dakejahl, 2019-10-23:
Using the `pwm test -p 1600 -a` works great, motors all spin without freaking out. I'll post a log of raw data once I am able to test with `master`, blocked by an erroneous  "heading estimate not stable, preflight checks failed" on `master` ... not sure why because if I flash `1.9.2` I can fly around with no issues.

MaEtUgR, 2019-10-24:
> "heading estimate not stable, preflight checks failed"

had it once as well, @bresch fixed it in https://github.com/PX4/Firmware/pull/13261

If it's changing from 1.9.2 to master it has to be a software caused problem. Currently I think the vibrations getting amplified is the most plausible one since I didn't have the fc mounted to the fram -> no vibrations and your pwm test also runs fine. You could try throttling up a bit with much lower `MC_(ROLL,PITCH,YAW)RATE_K` gain and check if your problem is gone. Be aware that you might not be able to fly in a stable fashion with much lower gains.

It would be really helpful to debug if you could like @jkflying suggested provide a high rate log and a short video showing how the symptom actually looks like such that we are sure it's there are no misunderstandings.

dakejahl, 2019-10-24:
@MaEtUgR I've got the FC isolated on a little vibe isolation mount with damping bulbs but I think you're correct. 

**1.9.2** 
_You can actually still see the effects of vibration on the motor output at high throttle. And I can indeed hear the bad sounds the motors make near high throttle._ 
[image]
https://review.px4.io/plot_app?log=7f7aba71-125c-4861-9b67-6ced2962f7a1

**master**
_Yup this is totally messed up. I am surprised I couldn't see this with low rate logging._
[image]
https://review.px4.io/plot_app?log=dccabbd4-65cd-4efd-bd10-421a9504dfb4

I guess my big question is: Why? Is this a consequence of running the attitude rate loop at a higher frequency? What can I do to fix this? Why am I the only person who has experienced this? (does everyone else have baller vibration damping? :wink: )

julianoes, 2019-10-25:
@dakejahl that looks like crazy vibrations on gyro and accel, wow! I don't understand where it comes from. Is there a chance you could do a `git bisect` to try to figure out when it was introduced?

dakejahl, 2019-10-25:
Yes I will investigate this early next week when I am back in town.

dakejahl, 2019-10-30:
Ha... of course the very first commit from `git bisect` doesn't even work. Can't arm, vehicle doesn't see RC.

julianoes, 2019-10-31:
@dakejahl you can do `git bisect skip` and try the next commit.

dakejahl, 2019-10-31:
Sad day... while testing during a `bisect`, the vehicle flipped itself and smashed on the ground destroying two props. Unfortunately I need to order some more, my spares are incompatible with these motors.
https://review.px4.io/plot_app?log=a7f25022-d98f-4cc4-b89b-12cda7d77760

dakejahl, 2019-10-31:
Here's a bench test with no props attached, I arm the vehicle and let it idle for a little bit, then ramp up the throttle to 0.5 and then back down to idle.
**1.9.2**
https://review.px4.io/plot_app?log=bf9f4248-a7e5-42dd-9f83-e60df970ad0a
**master**
https://review.px4.io/plot_app?log=eccaa7ec-6fb8-4e93-88dd-9689243c1c05

You can clearly see the raw gyro/accel go absolutely nuts on `master` as soon as there is more than idle throttle.

dakejahl, 2019-10-31:
Changing IMU_GYRO_CUTOFF from 80 back to 30 solved the issue :confused: 

I can't remember why I changed it away from default in the first place.... :man_shrugging: But still concerning that there is such a large difference in behavior between two versions.

https://review.px4.io/plot_app?log=b214a366-11a8-4ca3-9850-0faa3fd62c57

LorenzMeier, 2019-11-01:
That's not really concerning because we've changed the filtering pipeline to accommodate high-performance racers so that the cutoff setting is more effective. I'll let @bkueng / @bresch chime in, but that frame has a lot of flex and so what I'm seeing here is what I would expect with 30 vs 80 Hz cutoff.

mrpollo, 2019-11-03:
Thanks for doing the extra work to verify this @dakejahl.

@bkueng @bresch, please close this issue if you can verify @LorenzMeier conclusion, we need to move on with the v1.10 release.

bresch, 2019-11-03:
@mrpollo I think we still need to find the cause of the issue, lowering the cutoff just hides the underlying possible issue. 
This is why the bisect is really important to helping us finding the modifications that leads to the change of behavior.

@dakejahl any news?

dakejahl, 2019-11-03:
Unfortunately I can't continue the bisect with the s500 until I get some
new props. I'm going to setup my 250 airframe in the mean time and see if I
can't reproduce the issue. Travelling this week so I'll be back on it the
following week.

julianoes, 2019-11-05:
Here is another example of a frame with a lot of vibration and presumably overdriven motors:
https://discuss.px4.io/t/quad-crash-at-manual-mode/13579/3

closed 2020-01-09 by dakejahl