PX4/PX4-Autopilot #13148: QGC _altitude issues_CUAV Nano

Tony3dr, 2019-10-09:
**Describe the bug**
We have noticed the altitude issues on the CUAV Nano this issue was noted on QGC altitude reading. (and relative altitude) not being displayed correctly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Uploaded mission
3. Takeoff, Position Mode 
4. Switch from Position mode to Mission mode 
4. See altitude reading on QGC

**Log Files and Screenshots**
Log Nano V5:   
https://review.px4.io/plot_app?log=e3d7bc0a-891d-4ecf-8ebd-01afb7630ab7
[image]
[image]

**Drone (please complete the following information):**
FC- CUAV Nano 
Frame: F450 frame
GPS- NEO V2
[image]

**Additional context**
The vehicle flight performance is well but the altitude is off by approximately 10 to 13 meters.@dagar what could be the issue?

MaEtUgR, 2019-10-23:
I have multiple CUAV nano board here so I might be able to reproduce quickly.

Tony3dr, 2019-11-06:
> I have multiple CUAV V5 nano board here so I might be able to reproduce quickly.
@MaEtUgR were you able to reproduce the issue? Thanks

cuhome, 2019-11-11:
I tested it, but I didn't find the height of the error.
Logs:
Https://logs.px4.io/plot_app?log=19572b33-284d-457d-8028-a792349e8641

cuhome, 2019-11-11:
[image]
[image]

MaEtUgR, 2019-11-26:
I have a similar problem on a Pixhawk 4 mini setup. It was flying perfectly before and without any change at some point it began reading strange diverging altitude even when stationary on the desk and with freshly calibrated accelerometer.

I made a log from boot today with latest master when rebooting just after calibration: https://logs.px4.io/plot_app?log=d3d0d46b-b850-4225-97a6-812aee45ff56

MaEtUgR, 2019-11-26:
I quickly checked flashing 1.9.0 and I have the same problem. Let's analyze the log.

MaEtUgR, 2019-11-28:
Logs for comparison from the testing team's setup (thanks @Tony3dr !):
https://review.px4.io/plot_app?log=d8b837a4-1881-4895-b9b1-1df9008b5237
https://review.px4.io/plot_app?log=a5f8141a-6991-4217-8520-426b533bfff2

MaEtUgR, 2019-12-03:
Thanks to @kamilritz for looking into the logs. He found that the malicious drifting altitude in the tests on the bench come directly from the barometer... I wonder why certain boards have this a lot more than others. We should test if it's temperature dependent or see if the data sheet mentions some compensation.

MaEtUgR, 2020-03-12:
Is this problem still present? I'd assume this was fixed in the meantime because there were various related issues addressed e.g. accelerometer bias drifting https://github.com/PX4/ecl/pull/749, fusing height multiple times https://github.com/PX4/ecl/pull/767 and a lot more that I don't know of. The platform Holybro QAV250 with pixhawk 4 mini that I had problems with doesn't show them anymore. Please reopen if you see it again.

closed 2020-03-12 by MaEtUgR

Tony3dr, 2020-03-12:
> Is this problem still present?
> 
> I'd assume this was fixed in the meantime because there were various related issues addressed e.g. accelerometer bias drifting [PX4/ecl#749](https://github.com/PX4/ecl/pull/749), fusing height multiple times [PX4/ecl#767](https://github.com/PX4/ecl/pull/767) and a lot more that I don't know of. The platform Holybro QAV250 with pixhawk 4 mini that I had problems with doesn't show them anymore.
> 
> Please reopen if you see it again.

@MaEtUgR the issue can be closed, we haven't seen this issue anymore. Thanks for all the helop!