PX4/PX4-Autopilot #13153: Strange altitude behavior when EKF2_HGT_MODE = Range sensor

artifactz, 2019-10-10:
**Description**
Local position Z behaves unexpected on altitude changes. After a small delay (~1s) it seems to counteract the actual movement. Not influenced by `EKF2_RNG_GATE` or `EKF2_RNG_NOISE`.

**To Reproduce**
1. Set `EKF2_HGT_MODE` to `Range sensor`.
1a. Also disable EKF2_RNG_AID.
2. Lift drone manually.

**Expected behavior**
Local position Z goes from 0 to about -1.5m.

**Screenshots**
Drone was lifted, hold and put back down, this is how local position Z looks like:
[image]

Actual range finder values:
[image]

Also, I have a vision/optical flow module using the same range finder. If I publish its pose to `/mavros/vision_pose/pose_cov` and set `EKF2_HGT_MODE` to `Vision` the altitude behaves correctly (green/blue: x/y, red: z):
[image]

**Drone**
- quadcopter

**Additional context**
I was actually trying to feed in optical flow via MAVLink's `OPTICAL_FLOW_RAD` messages instead of MAVROS' `vision_pose` topic while I stumbled across this. Using the vision_pose topic works. Using the optical flow messages (while setting `EKF2_AID_MASK` appropriately of course) appears to exhibit similar behavior as the range finder.

julianoes, 2019-10-11:
Could you also add a logfile? thx.

artifactz, 2019-10-15:
@julianoes 
* EKF2_HGT_MODE = Range sensor
https://logs.px4.io/plot_app?log=61b7bce1-7ff9-4c58-b75d-beebc4d315d4
* EKF2_HGT_MODE = Vision
https://logs.px4.io/plot_app?log=58cd329c-b248-4de6-a323-a51fc8855c21

julianoes, 2019-10-17:
@RomanBapst could you check this? Thx.

MWBBMWID, 2019-10-29:
Are you using the px4flow integrated Sonar as the height source?

artifactz, 2019-10-31:
@farhangnaderi 
> Are you using the px4flow integrated Sonar as the height source?

No, it's a laser rangefinder.

bresch, 2021-01-27:
Closing as so many things changed in EKF2. Please open a new issue if this can be reproduced in master

closed 2021-01-27 by bresch