PX4/PX4-Autopilot #13155: Flip Mode

Seeelefant, 2019-10-10:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

**Describe your preferred solution**
A clear and concise description of what you want to happen.
Within PX4 I am missing a "Flip Mode" like in ArduCopter:
"...the Vehicle will flip on its roll or pitch axis depending upon the pilot’s roll and pitch stick position. Vehicle will rise for 1 second and then rapidly flip. The vehicle will not flip again until the switch is brought low and back to high."
Best
--Seeelefant

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots for the feature request here.

hamishwillee, 2019-10-11:
This is an open source project, which means that a mode like this will be developed if some member of the community sees it as valuable. If you want to create a PR to deliver the feature I'm sure we'd accept it :-)

mhkabir, 2019-10-11:
As Hamish said, we'll happily accept this feature if you decide to develop it and send a PR. Closing this for now.

closed 2019-10-11 by mhkabir

Seeelefant, 2019-10-14:
Thank you, I am not if I have time for such a project. Anybody willing to join?