PX4/PX4-Autopilot #13161: Add camera triggering in omnibus

Razbotics, 2019-10-11:
I am working on how to correctly enable and define pins for camera triggering for omnibus F4 SD.
Default it is not enabled and if enabled from cmake pin not defined error occurs. Please guide me the exact pin definition of 5 and 6 for omnibus F4 SD.

julianoes, 2019-10-11:
> error occurs.

Can you paste the actual error?

And can you give some more background on what you're trying to achieve and what exactly you have tried, and what docs you have looked at, etc.?

Razbotics, 2019-10-11:
The camera trigger parameters doesn't appears by default as the camera trigger driver is disabled to save flash since omnibus has 1MB memory. So what i have done is uncommented the camera_trigger in driver section in default.cmake in board/omnibus folder.
And then i ran  `make omnibus_f4sd_default` 
The error i get is this
[image]

After some digging into code and tracing the error I went into ~/Desktop/Firmware/boards/omnibus/f4sd/src/board_config.h
I uncommented 
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
`/* User GPIOs
 *
 * GPIO0-5 are the PWM servo outputs.
 * GPIO_TIM3_CH3OUT        GPIO_TIM3_CH3OUT_1 //PB0 S1_OUT D1_ST7
 * GPIO_TIM3_CH4OUT        GPIO_TIM3_CH4OUT_1 //PB1 S2_OUT D1_ST2
 * GPIO_TIM2_CH4OUT        GPIO_TIM2_CH4OUT_1 //PA3 S3_OUT D1_ST6
 * GPIO_TIM2_CH3OUT        GPIO_TIM2_CH3OUT_1 //PA2 S4_OUT D1_ST1
 * GPIO_TIM5_CH2OUT        GPIO_TIM5_CH2OUT_1 //PA1 S5_OUT
 * GPIO_TIM1_CH1OUT        GPIO_TIM1_CH1OUT_1 //PA8 S6_OUT
 */

#define _MK_GPIO_INPUT(def) (((def) & (GPIO_PORT_MASK | GPIO_PIN_MASK)) | (GPIO_INPUT|GPIO_PULLUP))

#define GPIO_GPIO0_INPUT             _MK_GPIO_INPUT(GPIO_TIM3_CH3IN)
#define GPIO_GPIO1_INPUT             _MK_GPIO_INPUT(GPIO_TIM3_CH4IN)
#define GPIO_GPIO2_INPUT             _MK_GPIO_INPUT(GPIO_TIM2_CH4IN)
#define GPIO_GPIO3_INPUT             _MK_GPIO_INPUT(GPIO_TIM2_CH3IN)
#define GPIO_GPIO4_INPUT             _MK_GPIO_INPUT(GPIO_TIM5_CH2IN) //Uncommented here
#define GPIO_GPIO5_INPUT             _MK_GPIO_INPUT(GPIO_TIM1_CH1IN) //Uncommented here

#define _MK_GPIO_OUTPUT(def) (((def) & (GPIO_PORT_MASK | GPIO_PIN_MASK)) | (GPIO_OUTPUT|GPIO_PUSHPULL|GPIO_SPEED_2MHz|GPIO_OUTPUT_CLEAR))

#define GPIO_GPIO0_OUTPUT            _MK_GPIO_OUTPUT(GPIO_TIM3_CH3OUT)
#define GPIO_GPIO1_OUTPUT            _MK_GPIO_OUTPUT(GPIO_TIM3_CH4OUT)
#define GPIO_GPIO2_OUTPUT            _MK_GPIO_OUTPUT(GPIO_TIM2_CH4OUT)
#define GPIO_GPIO3_OUTPUT            _MK_GPIO_OUTPUT(GPIO_TIM2_CH3OUT)
#define GPIO_GPIO4_OUTPUT            _MK_GPIO_OUTPUT(GPIO_TIM5_CH2OUT) //Uncommented here
#define GPIO_GPIO5_OUTPUT           _MK_GPIO_OUTPUT(GPIO_TIM1_CH1OUT) //Uncommented here
`
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

and further down
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
`#define GPIO_TIM3_CH3OUT        GPIO_TIM3_CH3OUT_1 //PB0 S1_OUT D1_ST7
#define GPIO_TIM3_CH4OUT        GPIO_TIM3_CH4OUT_1 //PB1 S2_OUT D1_ST2
#define GPIO_TIM2_CH4OUT        GPIO_TIM2_CH4OUT_1 //PA3 S3_OUT D1_ST6
#define GPIO_TIM2_CH3OUT        GPIO_TIM2_CH3OUT_1 //PA2 S4_OUT D1_ST1
#define GPIO_TIM5_CH2OUT        GPIO_TIM5_CH2OUT_1 //added this
#define GPIO_TIM1_CH1OUT        GPIO_TIM1_CH1OUT_1 //added this

#define DIRECT_PWM_OUTPUT_CHANNELS      4

#define GPIO_TIM3_CH3IN         GPIO_TIM3_CH3IN_1
#define GPIO_TIM3_CH4IN         GPIO_TIM3_CH4IN_1
#define GPIO_TIM2_CH4IN         GPIO_TIM2_CH4IN_1
#define GPIO_TIM2_CH3IN         GPIO_TIM2_CH3IN_1
#define GPIO_TIM5_CH2IN         GPIO_TIM5_CH2IN_1 //added this
#define GPIO_TIM1_CH1IN         GPIO_TIM1_CH1IN_1 //added this

#define DIRECT_INPUT_TIMER_CHANNELS  4
`
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Here is the complete file 
[board_config.h.txt](https://github.com/PX4/Firmware/files/3718621/board_config.h.txt)

Also in ~/Desktop/Firmware/boards/omnibus/f4sd/src/timer_config.c
I modified the file like this 
added PA1 S5_OUT entry and PA8 S6_OUT entry for the Trigger pins

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
`/*
 * OUTPUTS:
[... 43 lines cut ...]
      //PA1 S5_OUT
	{
		.gpio_out = GPIO_TIM5_CH2OUT,
		.gpio_in = GPIO_TIM5_CH2IN,
		.timer_index = 0,
		.timer_channel = 2,
		.ccr_offset = STM32_GTIM_CCR2_OFFSET,
		.masks  = GTIM_SR_CC2IF | GTIM_SR_CC2OF
	},
      //PA8 S6_OUT
	{
		.gpio_out = GPIO_TIM1_CH1OUT,
		.gpio_in = GPIO_TIM1_CH1IN,
		.timer_index = 0,
		.timer_channel = 1,
		.ccr_offset = STM32_GTIM_CCR1_OFFSET,
		.masks  = GTIM_SR_CC1IF | GTIM_SR_CC1OF
	}
};`
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Here is the complete file **Note that I have postfix .txt extension for uploading here**
[timer_config.c.txt](https://github.com/PX4/Firmware/files/3718632/timer_config.c.txt)

All these entries are just pattern matching from other four entries. I dont know anything at micro controller level so strongly suspect these things will make things wrong.
But what i noticed now the error was fixed and successfully built. I uploaded the code and tried camera triggering and the trigger works on camera but what i found is changing interval or trig_modes parameters doesn't change anything. The camera is always triggered at 31 Hz no matter what params i change. I think i did things in wrong way there are more things to modify and that needs in depth knowledge of timers and pins at micro controller level. Omnibus actually uses STM32 F405 RGT6 MCU.

julianoes, 2019-10-14:
Thanks for the info. FYI @bkueng @davids5

Razbotics, 2019-10-15:
> Thanks for the info. FYI @bkueng @davids5

I hope someone comes out and solve this issue. Though its not a bug just a feature needs to be added in omnibus F4 board. I will be glad if i contribute something.

davids5, 2019-10-17:
@Razbotics - without digging in on this what I can tell you is the PWM rate is set per TIMER. The Timer has channels and all timers channels run at the same rate.  

The above channel allocation looks incorrect. 

There are 2 timers. TIM2 and TIM3 with 2 channels each.
This is notes here correctly.
```
 *  M3 : PA3 : TIM2_CH3
 *  M4 : PA2 : TIM2_CH4
 *  M1 : PB0 : TIM3_CH3
 *  M2 : PB1 : TIM3_CH4
```
This is incorrect:
timer_index = 0 -> means that TIM2 is indexed.
```
      //PA1 S5_OUT
	{
		.gpio_out = GPIO_TIM5_CH2OUT,
		.gpio_in = GPIO_TIM5_CH2IN,
		.timer_index = 0,
		.timer_channel = 2,
		.ccr_offset = STM32_GTIM_CCR2_OFFSET,
		.masks  = GTIM_SR_CC2IF | GTIM_SR_CC2OF
	},
      //PA8 S6_OUT
	{
		.gpio_out = GPIO_TIM1_CH1OUT,
		.gpio_in = GPIO_TIM1_CH1IN,
		.timer_index = 0,
		.timer_channel = 1,
		.ccr_offset = STM32_GTIM_CCR1_OFFSET,
		.masks  = GTIM_SR_CC1IF | GTIM_SR_CC1OF
	}
```

Razbotics, 2019-10-17:
@davids5 Well i get that my timer configuration for pin 5 and 6 on omnibus is incorrect. The 4 PWM motor pins are correctly defined and i don't have to make any changes to them. I want the pin5 or pin6 to have triggering capabilities for camera, So for that i seek help for correcting pin configuration on omnibus board target.

davids5, 2019-10-18:
@Razbotics you will need a copy of the schematic to verify the wiering as to what pins are used. Then you will need to add the timer definitions (see px4\fmu-v5 for an example)  There will need to be 4 timers. Then the index of the channels need to point to timer index 2 and index 3. The Timers need to have the first and last index of the channels. First = last = 4 and First = last = 5 for the timers respectively. Then in board_config.h you will need to change DIRECT_PWM_OUTPUT_CHANNELS and DIRECT_INPUT_TIMER_CHANNELS  to 6

Razbotics, 2019-11-13:
@davids5 thanks for the info, and sorry for the late reply. I got what you want to say is that i have to dig into schematics for pinouts. But what is the line "First = last = 4 and First = last = 5 for the timers respectively" means. Can you please elaborate on that? It would be helpful.

Razbotics, 2019-12-04:
I have configured and activated all 6 PWM pins on Omnibus according to there TIMERS and CHANNELS mentioned in schematics. But still the camera trigger is not working properly. No matter what trigger params I change the camera triggers at 25Hz at all 6 pwm pins. Even if I disable the trigger in params, the camera continues to get triggered at 25HZ, Strange.
I am expecting some help here @davids5 @bkueng @mhkabir