PX4/PX4-Autopilot #13163: SITL gazebo typhoon_480 shows oscillations in flight

Jaeyoung-Lim, 2019-10-11:
**Describe the bug**
When running SITL gazebo with the `typhoon_480` oscillations can be observed during flight.

**To Reproduce**
1. Run stil using `make px4_sitl gazebo_typhoon_h480`
2. Upload a mission and start the mission

**Log Files and Screenshots**
The oscillations are quite obvious in this [log](https://review.px4.io/plot_app?log=b46dfced-2ea6-4aed-8e0d-2c43240e96a0) It seems to happen sometimes during forward flight.

**Additional context**
This is one of the only setups that a gimbal is integrated into SITL. It would be great to have a functional setup of the `typhoon_h480` model

Jaeyoung-Lim, 2019-10-15:
This seems to be a result of the tuning of the attitude controller on the model. Will try to tune the model in SITL

dayjaby, 2019-10-21:
Any updates? I use those values on current master:

x + MC_PITCHRATE_I [266,541] : 0.0500                                                                                                                                                        
x + MC_PITCHRATE_P [269,544] : 0.1000                                                                                                                                                        
x + MC_PITCH_P [270,545] : 6.0000
x + MC_ROLLRATE_I [275,550] : 0.1000
x + MC_ROLLRATE_P [278,553] : 0.1500
x + MC_ROLL_P [279,554] : 6.0000

Jaeyoung-Lim, 2019-10-21:
@dayjaby Thanks! Does the values fix the oscillation? 

@TSC21 This is a separate issue. I can fix it by retuning the vehicle, but I wanted to tune it really well until I can make a PR

dayjaby, 2019-10-21:
@Jaeyoung-Lim yes, it does. Feel free to create a PR with those.

PR #13246 (2019-10-23): Retune SITL model typhoon h480
**Describe problem solved by this pull request**
This Fixes #13163 where the typhoon_480 was showing oscillations is forward flight in SITL

**Describe your solution**
These are the retuned values of the typhoon h480 coming from the suggestion of @dayjaby

**Test data / coverage**
Log: https://review.px4.io/plot_app?log=7d5eae16-1c0e-4c42-8912-b9176f4d27e6
closed 2019-10-23 by TSC21