PX4/PX4-Autopilot #13164: Minimum angle of "Controller roll limit" is high

moreba1, 2019-10-11:
Hi

I will set FW_R_LIM to 25 degree but i not allowed to set it under 35 degree in qgc. also some gliders must not bank more than 30 degree . please change the FW_R_LIM min .

sfuhrer, 2019-10-11:
30° is an extremely low (max!) bank angle. I don't know of any system that should not be allowed to bank so little. 
If you want to have your flight with less tight turns than you should adapt turning radius and/or your route planning, and not this parameter.

Btw, you can still set any parameter in QGC to a value outside of the "normal" range, by clicking "enforce safe" or something similar in the advanced menu.

moreba1, 2019-10-11:
https://github.com/mavlink/qgroundcontrol/pull/7910

moreba1, 2019-10-11:
> Btw, you can still set any parameter in QGC to a value outside of the "normal" range, by clicking "enforce safe" or something similar in the advanced menu.

i can not force save this parameter

sfuhrer, 2019-10-11:
> > Btw, you can still set any parameter in QGC to a value outside of the "normal" range, by clicking "enforce safe" or something similar in the advanced menu.
> 
> i can not force save this parameter

I quickly checked, works for me.

closed 2019-10-12 by moreba1