PX4/PX4-Autopilot #13165: Thermal Compensation parameters are not categorized as system parameters

BazookaJoe1900, 2019-10-11:
**Describe the bug**
Its seems that the TC_xxx (Thermal Compensation) are not defined as system, but from my understood, parameters that categorized as 'system' are parameters that the user should not change manually, such as the calibration.
therefore they need to, right?

Is there manual that define exactly which parameters should be 'system'/'developer'/'volatile'? and how the user should handle them?

BazookaJoe1900, 2019-11-08:
@julianoes - how can answer to this?

hamishwillee, 2019-11-10:
I didn't think there was such a categorisation (though there has been much talk over the years). This is a note to me that if @julianoes says there is a way to do this that I need to document it.

julianoes, 2019-11-11:
I have no idea and no opinion about this.

LorenzMeier, 2019-11-11:
They should be clearly system parameters as they are factory level.

LorenzMeier, 2019-11-11:
After looking into it again, I think the param parsing needs to be updated.

BazookaJoe1900, 2019-11-11:
Thanks, I will PR that

LorenzMeier, 2019-11-11:
To be more verbose: What you're asking for requires extending QGC to not only hide System Level params, but also temperature calibration params, which are a different category.

BazookaJoe1900, 2019-11-11:
@LorenzMeier 
At this point I just want to define the TC_ parameters as system.
How does QGC reflect that parameter is a 'system'?

julianoes, 2019-11-11:
@BazookaJoe1900 I think like this:
https://github.com/PX4/Firmware/blob/07d656e971a72d1202651dfd3b4642736fb078d7/src/lib/systemlib/system_params.c#L48

BazookaJoe1900, 2019-11-11:
@julianoes Thanks. my main question is how it effect (and if) on QGC.

hamishwillee, 2019-11-11:
@BazookaJoe1900 My interpretation of the [above](https://github.com/PX4/Firmware/issues/13165#issuecomment-552360918) is that what you want to do is very much needed but not yet supported in QGC. So you can markup however you like, but parameters will not be hidden.

So what you would do is create a PR or issue in QGC such that any parameter with metadata type thermal calibration or System is hidden by default. What exactly "hidden" means should be discussed with Don, but given they are documented and visible in source, I'd suggest perhaps not so much "hidden" as "disabled for editing".

BazookaJoe1900, 2019-11-12:
@hamishwillee Thanks.
About how QGC handle that, is another issue.
because I have some other tool aht uses the 'system' configuration to decide if to deploy parameter or not, I wanted to fix this issue.
Closing this issue.

closed 2019-11-12 by BazookaJoe1900