PX4/PX4-Autopilot #13166: Pixhawk can not connect any computer

themystick, 2019-10-11:
Hi. I'm using Pixhawk 2.1 Cube. I have a problem with my device and I've tried to solve this problem many times, I've tried many things but still not working. So I need your help. The problem is my device is not connecting to any computer. When I tried to connect via micro usb port, the led (yellow and slowly blinked amber) is on but I can't see the driver in the device manager, I mean there is no difference in the device manager. So I've tried to connect via RFD and I can't connect again. (I've tried this with other computers). 

So interestingly, I've tried to connect with "debug" port and I've seen that the COM3 port is now available in the  device manager but the leds are off. Just connecting but It seems there is no power in the pixhawk. So I've tried to connect MP again and it's not working. 

So I've tried everything but It's not working. Please help me about this issue. Btw I've upload the video about the problem If you want . Thanks. Regards

julianoes, 2019-10-14:
> I've tried many things but still not working

What have you tried?

Are you trying to connect with QGC? If so what version of QGC? And what error (if any) is shown?

Is this just one Pixhawk Cube or multiple?

julianoes, 2020-02-12:
Closing because no answer.

closed 2020-02-12 by julianoes