PX4/PX4-Autopilot #13167: 【Vtol】The left and right ailerons have a large output when they are stationary on the ground.

xdwgood, 2019-10-12:
Use branch:  master https://github.com/PX4/Firmware/commit/803a71928f106ccefb453fc2bdf960fb22a63a89

1.Place a tailsitter vtol drone vertically on a level surface, qgc shows the angle of yaw is 90
2.Not arm, but drone left and right ailerons have large output

analysis：
I think it's because the yaw_sp angle is 0 when not arm, but the customer is randomly placed on the drone, so the actual yaw may not be 0, which causes the controller to output to the left and right ailerons.

[image]

xdwgood, 2019-10-12:
I picked up the drone selection until the yaw angle is 0, and the left and right ailerons will return to the middle position.

LorenzMeier, 2019-10-12:
Could you send a pull request for that?

xdwgood, 2019-10-14:
@dagar  I have verified that this reset caused the problem I described.

```
				// reinitialize the setpoint while not armed to make sure no value from the last mode or flight is still kept
				if (!_v_control_mode.flag_armed) {
					Quatf().copyTo(_mc_virtual_att_sp.q_d);
					Vector3f().copyTo(_mc_virtual_att_sp.thrust_body);
					Quatf().copyTo(_v_att_sp.q_d);
					Vector3f().copyTo(_v_att_sp.thrust_body);
				}
```

https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L415

I want to remove it, do you have any idea?

xdwgood, 2019-10-16:
fix : #13198

MaEtUgR, 2019-10-16:
Here's the first attempt of the proper solution: https://github.com/PX4/Firmware/pull/13210
But that one needs more corner-case testing.

xdwgood, 2019-11-09:
@MaEtUgR  The latest master branch, this feature is broken again
https://logs.px4.io/plot_app?log=8c7b8d4e-42b7-4fa9-9ed7-5d364a133f19

xdwgood, 2019-11-10:
see here： https://github.com/PX4/Firmware/pull/13210

closed 2019-11-10 by xdwgood