PX4/PX4-Autopilot #13168: Derivative gain for fixedwing attitude control

moreba1, 2019-10-12:
Hi

There is need for derivative gain for fixedwing attitude control for tailess planes .

https://github.com/PX4/Firmware/issues/12886

julianoes, 2019-10-14:
Please create an issue with some background information, some suggestions how something could be done, what sort of problem you're trying to solve, etc.. With just this one sentence I don't know what to do with this.

And "there is need" doesn't mean anything will just create that for you, so if you need it soon, it's best to create a pull request. Thanks.

closed 2019-10-14 by julianoes

RomanBapst, 2019-10-14:
@moreba1 I think I understand what you are trying to say. But you should show some evidence first, e.g. plot for rate vs rate setpoints form this vehicle. Maybe we can suggest some changes in tuning parameters.