PX4/PX4-Autopilot #13172: Unable to build px4 current version by cross-compilation on macOS Mojave 10.14.6

jshuadvd, 2019-10-13:
Whenever I run the `make emlid_navio2_cross` command in attempt to build the current master branch of the code, I receive the following errors.

```
Firmware ⇨ ⇨ ⇨  △ △ △  👾  ⚡  || master || ⚡  👾   ⇨ ⇨ ⇨  make emlid_navio2_cross
-- PX4 version: v1.10.0-beta2-37-gc8f3f07ff7
-- PX4 config file: /Firmware/boards/emlid/navio2/cross.cmake
-- PX4 config: emlid_navio2_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: arm-linux-gnueabihf-gcc
CMake Error at CMakeLists.txt:189 (project):
  The CMAKE_CXX_COMPILER:

    arm-linux-gnueabihf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:189 (project):
  The CMAKE_C_COMPILER:

    arm-linux-gnueabihf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:189 (project):
  The CMAKE_ASM_COMPILER:

    arm-linux-gnueabihf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also `/Firmware/build/emlid_navio2_cross/CMakeFiles/CMakeOutput.log`.
See also `/Firmware/build/emlid_navio2_cross/CMakeFiles/CMakeError.log`.
Error: `/Firmware/build/emlid_navio2_cross` is not a directory
make: *** [emlid_navio2_cross] Error 1

```

I'm not 100% sure what seems to be going on here, but I can only build with `make px4_sitl jmavsim` not gazebo either. Any help would be greatly appreciated. Thank you!

julianoes, 2019-10-14:
You are trying to cross-compile without the cross compilation toolchain available. You need gcc that can compile for RPi.

```
CMake Error at CMakeLists.txt:189 (project):
  The CMAKE_CXX_COMPILER:

    arm-linux-gnueabihf-g++

  is not a full path and was not found in the PATH.
```

jshuadvd, 2019-10-14:
@julianoes thank you for the response! 

Is there a proper way to set this up on the mac that you would recommend ? 

It seems that the most recent [post](https://medium.com/@haraldfernengel/cross-compiling-c-c-from-macos-to-raspberry-pi-in-2-easy-steps-23f391a8c63) I can find is over a year old.

julianoes, 2019-10-15:
You need to have the toolchain (`arm-linux-gnueabihf-g++`) somewhere and then make sure the `bin` folder with the binaries is in your path, e.g.:
```
export PATH=$PATH:/path/to/the/toolchain
```

jshuadvd, 2019-10-15:
@julianoes thank you again. 

Sorry to be so annoying, but I am clearly missing something here. 

I have the toolchain installed that comes built with the homebrew `brew tap PX4/px4` 
(Please see the image below.) 
[image]

Are you saying that I need to have an additional toolchain installed on my Mac as well?

Or that I need to add the existing tapped repo compiler location to the `$PATH` variable?

Again, sorry about the confusion. I appreciate your help!

julianoes, 2019-10-15:
It should all be covered here actually: https://dev.px4.io/master/en/setup/dev_env_mac.html

jshuadvd, 2019-10-15:
@julianoes thanks, but I have followed the instructions to a t, over and over again and this is why I have posted.  I can’t seem to find any other help. 

Let me back up a little to explain. 

After following all of the instructions that you referenced, I am able to build with `make px4_sitl jMAVSim` but cannot use gazebo or the Emlid navio2 command which is what I need to do. 

The aforementioned errors are from when trying to compile for the navio2. Before even exporting the file to upload to the pi, I need to compile it first. I can’t do that because of said errors. 

Any further help would be greatly appreciated.

julianoes, 2019-10-15:
Right, sorry. I'm not entirely sure which toolchain you need to cross compile for RPi.

PX4 does provide a docker image though with the toolchain installed, however, the toolchain itself is based on GCC 4.8.2 and 4 years old.

You can use it using docker:
```
Tools/docker_run.sh make emlid_navio2_cross
```

This is the dockerfile for it:
https://hub.docker.com/r/px4io/px4-dev-raspi/dockerfile

jshuadvd, 2019-10-15:
Hmm. Ok. Thank you. 

I assume that the toolchain that comes with the brew tap for macOS should work automatically theoretically, especially since it does compile for jMAVSim. And in all of my research and reading (even on the ARM site), the toolchain installed is the correct one to cross compile for any environment from the Mac. 

Should this be considered a bug instead of a question ? As that info is not mentioned in the documentation.

hamishwillee, 2019-10-15:
> Should this be considered a bug instead of a question ? As that info is not mentioned in the documentation.

MacOS very clearly supports only NuttX, JMavsim, Gazebo:
- http://dev.px4.io/master/en/setup/dev_env.html#supported-targets
- https://dev.px4.io/master/en/setup/dev_env_mac.html
  [image]

If emlid_navio2_cross is one of those then this is a bug, if not it is is a question.

jshuadvd, 2019-10-15:
Thank you @hamishwillee I understand that. The issue is cross compilation from macOS to raspberry pi for the emlid_navio2_cross. 

Not macOS itself as you can see from the thread above.

hamishwillee, 2019-10-15:
@jshuadvd You are on macOS and you want to target raspberry pi. RaPi is not listed as a supported target that you can build for on macOS. I don't understand the distinction that you are trying to draw.

jshuadvd, 2019-10-16:
@hamishwillee I see what you are saying. I was under the impression that since I am not building for  macOS specifically, I could cross-compile (since this is the recommended way to build vs native) for the the pi and then ssh to upload as per [this](https://dev.px4.io/master/en/setup/building_px4.html) link. 

As it seems like a stepped process, I just followed along, and after building successfully with `jmavsim` jumped down to [this](https://dev.px4.io/master/en/setup/building_px4.html#raspberry-pi-23-boards) section. 

Similarly, it doesn't say anywhere specifically that you cannot cross compile from osx to rpi. And I have seen a few videos where individuals are doing just that. 

To clarify, are you saying that the only way to build for the navio2 is on a linux machine? Thank's for your help!

hamishwillee, 2019-10-16:
@jshuadvd 

> To clarify, are you saying that the only way to build for the navio2 is on a linux machine? 

Nearly. I am saying that the only _supported_ way to build for the navio2 is on a Linux machine. It may well be possible to build on MacOS, but we don't have a supported solution.

What does supported mean? Essentially, a system that is known to work, that is tested in CI and that we commit to keeping working.

You're right that we don't say you can't cross compile from OSX to RPi. What we do is say what you can compile. That makes more sense, because after all the list of things you can't compile to is very large.

julianoes, 2019-10-17:
Have you tried the docker solution? This should work on macOS.

jshuadvd, 2019-10-18:
@julianoes I haven't tried yet. I plan to do this tonight and see if that gets me anywhere. :(

jshuadvd, 2019-10-23:
@julianoes Docker did not work for me either unfortunately.

After installing the containers  (see image)

[image]

following the instructions listed on the documentation, and entering this command:

`sudo ./Tools/docker_run.sh 'make px4_sitl_default'`

below is the error I receive. 

```
guessing PX4_DOCKER_REPO based on input
PX4_DOCKER_REPO: px4io/px4-dev-nuttx:2019-07-29
Starting with UID : 0
usermod: UID '0' already exists
No such file or directory
CMake Error: Generator: execution of make failed. Make command was: "/Users//anaconda3/bin/ninja"
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
```

julianoes, 2019-10-25:
Don't run it with `sudo`!

jshuadvd, 2019-10-26:
Please see the updated error:
```
Firmware ⇨ ⇨ ⇨  △ △ △  👾  ⚡  || master || ⚡  👾   ⇨ ⇨ ⇨  ./Tools/docker_run.sh 'make px4_sitl_default'
guessing PX4_DOCKER_REPO based on input
PX4_DOCKER_REPO: px4io/px4-dev-nuttx:2019-07-29
Starting with UID : 502
No such file or directory
CMake Error: Generator: execution of make failed. Make command was: "/Users/joshuadavid/anaconda3/bin/ninja"
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
Firmware ⇨ ⇨ ⇨  △ △ △  👾  ⚡  || master || ⚡  👾   ⇨ ⇨ ⇨
```

and if i run: `./Tools/docker_run.sh 'make emlid_navio2_cross'`
this is the output:

```
Firmware ⇨ ⇨ ⇨  △ △ △  👾  ⚡  || master || ⚡  👾   ⇨ ⇨ ⇨  ./Tools/docker_run.sh 'make emlid_navio2_cross'
guessing PX4_DOCKER_REPO based on input
PX4_DOCKER_REPO: px4io/px4-dev-raspi:2019-07-29
Starting with UID : 502
-- PX4 version: v1.10.0-beta3-29-g070d75496d
-- PX4 config file: /Users/joshuadavid/Documents/Repositories/src/Firmware/boards/emlid/navio2/cross.cmake
-- PX4 config: emlid_navio2_cross
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 4.8.3
-- The C compiler identification is GNU 4.8.3
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler: /usr/lib/ccache/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/lib/ccache/arm-linux-gnueabihf-g++ -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "/usr/lib/ccache/arm-linux-gnueabihf-g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/joshuadavid/Documents/Repositories/src/Firmware/build/emlid_navio2_cross/CMakeFiles/CMakeTmp

    Run Build Command:"/usr/bin/ninja" "cmTC_c1334"
    [1/2] Building CXX object CMakeFiles/cmTC_c1334.dir/testCXXCompiler.cxx.o
    FAILED: CMakeFiles/cmTC_c1334.dir/testCXXCompiler.cxx.o
    /usr/lib/ccache/arm-linux-gnueabihf-g++     -o CMakeFiles/cmTC_c1334.dir/testCXXCompiler.cxx.o -c testCXXCompiler.cxx
    ccache: error: Failed to create directory /Users/joshuadavid/.ccache/tmp: Permission denied
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:189 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/joshuadavid/Documents/Repositories/src/Firmware/build/emlid_navio2_cross/CMakeFiles/CMakeOutput.log".
See also "/Users/joshuadavid/Documents/Repositories/src/Firmware/build/emlid_navio2_cross/CMakeFiles/CMakeError.log".
Error: /Users/joshuadavid/Documents/Repositories/src/Firmware/build/emlid_navio2_cross is not a directory
Makefile:193: recipe for target 'emlid_navio2_cross' failed
make: *** [emlid_navio2_cross] Error 1
Firmware ⇨ ⇨ ⇨  △ △ △  👾  ⚡  || master || ⚡  👾   ⇨ ⇨ ⇨
```

LorenzMeier, 2019-10-26:
You need to reset all directories that you've touched with the sudo command to your user (or delete them). You have now files that are owned by the super user.

I will close this issue - the things we're discussing here (like Mac OS file permissions) are trivial system level issues that you can read up on sites like Stackoverflow and this is not a support forum. Please post to the PX4 user forum and post the link here and some might still give you pointers.

But what you're trying is unsupported, so it is not an issue. We have a very small team and that team needs to focus its energy - so even when something seems possible (like running cross-compilation on Mac OS) we still might not spend time on it.

If you want to contribute documentation or code changes that make this easier for other Mac OS users, please send pull requests. I'm running on Mac OS myself and I would love that - but I still don't do it, because I can spend the full year dealing with toolchain issues on different operating systems.

Which is why we've chosen a supported set and leave it there. Thanks!

closed 2019-10-26 by LorenzMeier

jshuadvd, 2019-10-27:
Hi @LorenzMeier Thanks for your feedback. I was simply following the directions in the [documentation](https://dev.px4.io/master/en/test_and_ci/docker.html#helper-script-dockerrunsh) which clearly states to use the `sudo` command. 

I understand that this is not a support forum, and have posted to the discuss channel without any help as creating the issue was a last resort. 

To provide you with a little context, my project is used for a prototype. The reason I find PX4 interesting is the specific open source license type for Intellectual Property. I have no issue using AP (which is fully supported by the Navio2), but would rather use PX4. Perhaps I should look into Auterion for a more custom / supported solution. 

I will continue on my path, and gladly document any success i have. Again, thank you for your remarks as well as your great contribution to the world :) Have a wonderful rest of your weekend !

julianoes, 2019-10-28:
Thanks for pointing to the documentation! We'll fix that.

julianoes, 2019-10-28:
Fix for sudo: https://github.com/PX4/Devguide/pull/881

@LorenzMeier should we add a note that RPi with navio2 is not officially supported to prevent confusion?
https://docs.px4.io/v1.9.0/en/flight_controller/
https://docs.px4.io/v1.9.0/en/flight_controller/raspberry_pi_navio2.html
https://dev.px4.io/master/en/setup/building_px4.html#raspberry-pi-23-boards

LorenzMeier, 2019-10-28:
@bkueng To chime in - he worked on that a bit and I don't know where he left it off.

bkueng, 2019-10-28:
This builds fine for me (on Linux), using `arm-linux-gnueabihf-g++ (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3 20140303 (prerelease)`.

@jshuadvd your output points to a setup issue: you need `arm-linux-gnueabihf-g++` in your PATH, and you have to install that toolchain separately. You can check with `which arm-linux-gnueabihf-g++` if it's there.