PX4/PX4-Autopilot #13177: Time UTC usec goes backward

tuloski, 2019-10-14:
Analyzing a log I noticed that the field vehicle_gps_position_0.time_utc_usec goes backward of around 1 second, causing a "misalignment" of 2 seconds.

I have no idea if that is due to PX4 logger, to uOrb delay or to GPS data being erroneously calculated. I'm pretty sure it's not the GPS itself sending wrong data.

Any idea?

I can't share the log, but below is the image of the data on FlightPlot.

[image]

julianoes, 2019-10-14:
I think this expected. GPS is estimating position and UTC time. It makes sense that not just the position but also the time will get updated as a GPS signal improves.

tuloski, 2019-10-14:
It might be, but a 2s error is HUGE for GPS, so I'm not sure that is the cause.
2s error has the same order of magnitude of 1Km positioning error.

closed 2019-10-14 by tuloski