PX4/PX4-Autopilot #13192: Rover position control overrides offboard ActuatorControl

jbeyerstedt, 2019-10-15:
Setting actuator control values with MAVSDK does not affect the actual PWM outputs with a rover. When using QGC to monitor the raw PWM values, the PWM value is only set correctly in one MAVLink message, but mostly it's 0 (PWM value of 1500 us).

However, when the `rover_pos_control` module ist stopped, the offboard control affects the PWM values as expected.

(Setting Attitude set points (with [this MAVSDK-Python example](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/offboard_attitude.py)) wasn't working as expected either. But I haven't taken a further look at that problem.)

Jaeyoung-Lim, 2019-10-20:
@jbeyerstedt I think this is because the rover position control does not handle the case where the actuator control is enabled.

jbeyerstedt, 2019-10-21:
@Jaeyoung-Lim Thats also what I got from the rover call.

I will try to add support for offboard actuator_controls this week. Can you assign me?

I'll open another issue for attitude set points, because that will be a bigger issue. We might also need some help/ hints there.

commit 4ce03dfc (2019-11-20): Add support for offboard ActuatorControl set points to RoverPositionControl (fixes #13192) (#13314)

* fix formatting

* RoverPositionControl: Support Actuator Control Setpoints (fixes #13192)

* RoverPositonControl: remove control modes, that aren't currently implemented

* RoverPositionControl: use new Publication API
closed 2019-11-20 by julianoes