PX4/PX4-Autopilot #13195: Command retransmissions are denied

julianoes, 2019-10-15:
**Describe the bug**
When I try to start a mission in QGC but the vehicle does not have GPS yet, it will beep unhappy 3 times.

**To Reproduce**
Steps to reproduce the behavior:
1. Start HITL (or a real drone inside)
2. Upload a mission and try to start it before GPS lock (EKF2 position initialized)
3. Listen to 2 or 3 unhappy beeps

**Expected behavior**
It should only beep once per request.

**Additional context**
Presumably QGC retransmits the MAVLink command 3 times. It is not clear if the problem is that QGC sends the command 3 times without waiting for an ack or if PX4 doesn't or doesn't respond in time.

dagar, 2019-10-16:
Is this via the start mission slider in QGC?

julianoes, 2019-10-17:
> Is this via the start mission slider in QGC?

Yes it is.

dagar, 2019-10-17:
Not sure if this is related, but I believe QGC is sending separate commands for changing to MISSION, then arming, and possibly setting the first waypoint (first). 

I added MAV_CMD_MISSION_START (https://mavlink.io/en/messages/common.html#MAV_CMD_MISSION_START) as a possible replacement that does all together PX4 side, but QGC wasn't updated.

dagar, 2019-10-17:
QGC pull requests still open.
https://github.com/mavlink/qgroundcontrol/pull/5732
https://github.com/mavlink/qgroundcontrol/pull/5741

julianoes, 2019-10-18:
I don't think I like `MAV_CMD_MISSION_START`. It's seems redundant or overlapping existing commands. For instance it sets start and end but we already have [MISSION_SET_CURRENT](https://mavlink.io/en/messages/common.html#MISSION_SET_CURRENT).

dagar, 2019-10-18:
MISSION_SET_CURRENT isn't great because it causes the mission to reload entirely and rerun the feasbility checker. The benefit of `MAV_CMD_MISSION_START` is being able to do all thing atomically from the GCS.

julianoes, 2019-10-19:
> MISSION_SET_CURRENT isn't great because it causes the mission to reload entirely and rerun the feasbility checker.

I don't understand how that should be different. For both cases you potentially need to do the same checks.

dagar, 2019-10-19:
### MISSION_SET_CURRENT
 - standalone mavlink message, not a command (no acknowledge, retry, etc)
 - handled by the mavlink module in PX4 which sets the current index, writes back to dataman, publishes a new `mission` over ORB, then navigator mission has to reload the mission as if it was new, re-run feasibility across the entire thing again, etc.

### MAV_CMD_MISSION_START
 - single mavlink command
 - accepted if
    - mission valid (the feasibility status of the existing mission)
    - you've requested a valid index to start at
    - switching to mission mode succeeds
    - arming succeeds
 - asks navigator mission to simply update the current index of the mission
 - also used internally by VEHICLE_CMD_DO_LAND_START

There's not really a mechanism for QGC to be directly aware of the mission feasibility.