PX4/PX4-Autopilot #13201: uavcan_generic_driver error after make px4_fmu-v3_default

mcelhennyi, 2019-10-15:
Continuing from [here](https://github.com/PX4/Firmware/issues/12095?_pjax=%23js-repo-pjax-container)

I tried @hamishwillee's suggestion. I still have the same error. 

I forgot to mention in my post there, that the issue now when I try to make again is that I get this error:
```
user@Ian-laptop:~/gazebo_dev/px4-firmware$ make px4_fmu-v3_default
-- PX4 config file: /home/user/gazebo_dev/px4-firmware/boards/px4/fmu-v3/default.cmake
-- PX4 config: px4_fmu-v3_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.2
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 4.9.3
-- The C compiler identification is GNU 4.9.3
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Found PY_jinja2: /usr/local/lib/python3.5/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- Building and including px4_io-v2_default
-- Using C++11
-- Release build type: MinSizeRel
CMake Error at cmake/px4_add_module.cmake:210 (get_target_property):
  get_target_property() called with non-existent target
  "uavcan_generic_driver".
Call Stack (most recent call first):
  src/drivers/uavcan/CMakeLists.txt:70 (px4_add_module)

-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- Configuring incomplete, errors occurred!
See also "/home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default/CMakeFiles/CMakeOutput.log".
See also "/home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default/CMakeFiles/CMakeError.log".
Error: /home/user/gazebo_dev/px4-firmware/build/px4_fmu-v3_default is not a directory
Makefile:193: recipe for target 'px4_fmu-v3_default' failed
make: *** [px4_fmu-v3_default] Error 1

```

julianoes, 2019-10-17:
FYI @pavel-kirienko

OskarPepryk, 2020-09-23:
Just for people searching for solution to this problem. 
Similar problem appears when building old PX4 versions in Ubuntu 20.04 where default python interpreter is changed to python3. Solution is to install python2, pip for python2 and dependencies from Tools/setup/ubuntu.sh directory. Using update-alternatives map python cmd to python2. This fixed my problem with "uavcan_generic_driver".

closed 2026-08-11 by github-actions[bot]