PX4/PX4-Autopilot #13207: Maxbotix sonar sensor not working

tijmen-garuda, 2019-10-16:
Hi, 

I have trouble with connecting my Maxbotix MB1222 sonar to my holybro Pixhawk 4 running the latest stable version of PX4 (v.1.9.2). I have tried a lot, and basically it doesn't work. 

I connect it through the I2C A port, and all the cables are connected correctly. Earlier I also connected a Terabee V1 optical sensor through the same I2C A port, with which I could successfully read the `M1: DISTANCE_SENSOR`values in the "analyze" widget. Right now it (the `M1: DISTANCE_SENSOR`) does not show up however, even though I have both started mb12xx in mavlink manually, and changed the `SENS_EN_MB12XX` to "enabled" (which should start the driver if I understand correctly). 

When I go to the Mavlink console and run `mb12xx info` or `mb12xx stop` it keeps giving errors such as "driver poll restart failed" or "driver not running".  When looking in the /dev/ folder through mavlink however, I saw that there wasn't even a mb12xx folder driver present! 

Note: when I run `mb12xx start` it doesn't give any errors or respond. 

Does anyone know how to resolve this issue? And is the Maxbotix still supported by PX4? (the driver used to be present, but not anymore it seems?)

Thank you in advance,

Tijmen

mcsauder, 2019-10-16:
Hi @tijmen-garuda , it's worth a quick double check that the sticker on the can on the back of your sensor is 1222 and not 1220. (I was mistakenly sent a PWM device and not the i2c version in an order earlier this year and thought my sensor was broken when in reality it was the wrong communication type.)

I don't have an mb1222, but I checked my mb1202's and they are functioning as expected.  Are you running v1.9.2 stable, or current master?

Here is what I see from current master with 2 sensors on a pixhawk 4, i2c B port, (notice that `-a` is required to tell the driver to try all busses):
```
nsh> mb12xx start
MB12xx on I2C bus 1 at 0x70 (bus: 100 KHz, max: 100 KHz)
ERROR [mb12xx] no sensors discovered
nsh> mb12xx stop
INFO  [mb12xx] driver stopped
nsh> mb12xx start -a
MB12xx on I2C bus 1 at 0x70 (bus: 100 KHz, max: 100 KHz)
ERROR [mb12xx] no sensors discovered
MB12xx on I2C bus 2 at 0x70 (bus: 100 KHz, max: 100 KHz)
INFO  [mb12xx] sensor 1 at address 0x70 added
INFO  [mb12xx] sensor 2 at address 0x6F added
INFO  [mb12xx] Total sensors connected: 2
INFO  [mb12xx] driver started
nsh> 
```

mcsauder, 2019-10-16:
Hi @tijmen-garuda , my apologies, I see in your post that you are running v1.9.2.  Could you try current master branch?

tijmen-garuda, 2019-10-18:
Mcsaunder, thanks for your reply. I double checked the Maxbotix sticker on the backside of the sensor, and it is definitely a MB1222. I also flashed my drone with the current master branch as you suggested, but unfortunately I still cannot see any `M1:DISTANCE_SENSOR` values. Also I tried the same commands in mavlink console, which gave the following output:

[image]

The wiring is connected correctly, I also checked it with a multimeter. I am starting to think that the specific sensor type is either not being supported, or it must be broken somehow (though no damage is visible it might be internal on the board?). 

Any more suggestions are more than welcome.

mcsauder, 2019-10-18:
@tijmen-garuda , if you have access to a signal alayzer it would be easy to check if the device is answering back at anything at all.  I ordered an mb1222 so that I can try it on my end.  There might be something amiss with your sensor, I'll report back as soon as I have a unit to test with!

tijmen-garuda, 2019-10-20:
Hi mcsauder, unfortunately I do not have a signal analyzer around, but I do plan on connecting the mb1222 to one of the arduino's or raspberry pi's I have tomorrow. I'll keep you posted, and thank you for also ordering a mb1222! Very curious to find what your results are.

tijmen-garuda, 2019-10-21:
Hi all, we just connected the MB1222 to an adruino nano, but we cannot get it to work there either (we have followed all the regular setups & manuals to do that). The MB1222 is probably broken, though everything looks fine. Mcsaunder, could you let us know if yours does work?

mcsauder, 2019-10-21:
Hi @tijmen-garuda , my mb1222 showed up today and I was able to set the i2c address and run it alongside the other two mb1202 sensors I have:

[image]

I can also see 3 sensors reporting back distinct values on the bench in the analyze widget:
[image]

I think something is wrong with your sensor. Let me know if you have any questions, and otherwise just let me know if we can close out this issue.  Thanks!

tijmen-garuda, 2019-10-23:
Hi @mcsauder , I also ordered another batch of mb1222, an got the first one today. It seems this one does work, so the last one was definitely broken. Thanks for the help !

mcsauder, 2019-10-23:
Glad to hear your new sensor works!  I'll go ahead and close this issue!

closed 2019-10-23 by mcsauder