PX4/PX4-Autopilot #13209: [HIL VTOL Offboard] set_attitude() not working

ThomasRigi, 2019-10-16:
**Describe the bug**
Setting offboard attitude setpoints doesn't work for VTOLs in HIL simulation. The last attitude setpoint from before sending atttitude setpoints is kept instead.

This problem persists even with @irsdkv's fix #13141, even though now a few of the setpoints are registered : 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to branch of PR #13141 (optional, but recommended)
2. Start HIL simulation with the standard VTOL model (Gazebo)
3. Launch MAVSDK offboard example.
4. Observe the problem

**Expected behavior**
The vehicle should follow the offboard setpoints, like in this log where the MAVSDK offboard example works with Iris quadcopter in HIL: 
[https://logs.px4.io/plot_app?log=f032c92d-ffa3-4831-a730-977bbb8e229e](https://logs.px4.io/plot_app?log=f032c92d-ffa3-4831-a730-977bbb8e229e)

**Log Files and Screenshots**
Standard VTOL, HIL:
[https://logs.px4.io/plot_app?log=1fae08b3-82f3-4a45-b5d9-ff997bb9e7a3](https://logs.px4.io/plot_app?log=1fae08b3-82f3-4a45-b5d9-ff997bb9e7a3)

**Drone (please complete the following information):**
- HIL Standard VTOL QuadPlane, sys_autostart = 1002
- Pixhawk 4 is the used hardware

**Additional context**
#13141 fixes the problem for SITL, but interestingly enough not for HIL. Without #13141, it makes sense that it's not working, whereas with it I really don't know why the old setpoint persists.

LorenzMeier, 2019-10-17:
Thanks for reporting! Given this is a fairly specialized and new usage scenario and not affecting a lot of users this is something that will take a while to get back to for the core maintenance team. It would be appreciated if you could dive into it and send a pull request.

ThomasRigi, 2019-10-17:
I agree that it is something that will most likely not affect many users - for as long as this is not a hardware problem that ports over to real drones. (Even then it would be quite a limited user circle, but the consequences would be worse.) If real drones aren't affected and SITL is working, this issue certainly is on low priority - also for myself.

I tried to delve into it and thought to have tracked down every part of the code where something is written into the corresponding memory, but I couldn't find anything. Also, I struggled with using VSC's debugger for HIL simulation. Sometimes it kind of worked, but most of the time the HIL drone would refuse arming and again I couldn't figure out why. So don't count on a PR from me in the close future, sorry. :/

julianoes, 2019-10-17:
FYI @Jaeyoung-Lim