PX4/PX4-Autopilot #13220: Loiter to alt function

moreba1, 2019-10-17:
**Describe the bug**
loiter to alt : Do not give elevation change before reaching the loiter point.

**Expected behavior**

changing altitude during going to loiter point .

[image]

julianoes, 2019-10-17:
Does this param have an effect?
https://dev.px4.io/v1.9.0/en/advanced/parameter_reference.html#MIS_ALTMODE

dagar, 2019-10-17:
Where are you expecting it to intersect?

moreba1, 2019-10-17:
@julianoes  I dont know .

julianoes, 2019-10-18:
I'm not sure what the behaviour should be. What do you think @dagar?

dagar, 2019-10-18:
This is the intended behavior (using loiter to attain an altitude), but variations are certainly possible if @moreba1 can elaborate.

moreba1, 2019-10-18:
@dagar 
In my proposed behavior there is improvement in power saving and time saving in both descent and climb. At the end of flight in descent no need to motor power for going to loiter waypoint . also In starting flight after takeoff or transition to fw it is better and safer that planes gain height.

dagar, 2019-10-18:
That makes sense, I'm not sure that should be the behaviour within LOITER_TO_ALT (https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TO_ALT).

We should already be able to make this the norm within a regular waypoint or loiter.

moreba1, 2019-10-19:
The relevant code is in navigator/mission_block.cpp . In ardupilot code loiter_to_alt mode works like the proposed method. 

[image]

dagar, 2019-10-19:
LOITER_TO_ALT was specifically added for a landing sequence consisting of loitering down from the current altitude, then breaking out of the loiter on a tangent to begin your landing approach. Only descending via loiter is the point. 

Questions
1. How will your suggestion to change LOITER_TO_ALT impact that?
2. Why do you need LOITER_TO_ALT to behave this way if it can already be achieved with a regular loiter or waypoint?

moreba1, 2019-10-20:
``

> Why do you need LOITER_TO_ALT to behave this way if it can already be achieved with a regular loiter or waypoint?
regular waypoint can not be used because it has not "heading wait" .
can we define a waypoint in mission before LOITER_TO_ALT in same position ?

moreba1, 2019-10-27:
we can define initial altitude for loiter_to_alt points .

closed 2020-11-29 by moreba1