PX4/PX4-Autopilot #13223: make px4_sitl gazebo not working on macOS catalina

alexander-wu-aw, 2019-10-18:
When I followed the tutorial on the px4 developer website, everything worked perfectly. However, after I updated my computer to macOS Catalina - which then caused me to have to reinstall anaconda - the `make px4_sitl gazebo` command fails. 

I'm running macOS Catalina 10.15 (19A602)

This is the output after executing the command:

`[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
CMake Warning at /usr/local/lib/cmake/boost_system-1.71.0/libboost_system-variant-shared.cmake:59 (message):
  Target Boost::system already has an imported location
  '/usr/local/lib/libboost_system-mt.dylib', which will be overwritten with
  '/usr/local/lib/libboost_system.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_system-1.71.0/boost_system-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:33 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake:59 (message):
  Target Boost::filesystem already has an imported location
  '/usr/local/lib/libboost_filesystem-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_filesystem.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_filesystem-1.71.0/boost_filesystem-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:33 (find_package)

-- Found Boost: /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components:  system thread filesystem 
CMake Warning at /usr/local/lib/cmake/boost_regex-1.71.0/libboost_regex-variant-shared.cmake:59 (message):
  Target Boost::regex already has an imported location
  '/usr/local/lib/libboost_regex-mt.dylib', which will be overwritten with
  '/usr/local/lib/libboost_regex.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_regex-1.71.0/boost_regex-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  /usr/local/share/dart/cmake/DARTFindBoost.cmake:24 (find_package)
  /usr/local/share/dart/cmake/dart_dartComponent.cmake:6 (include)
  /usr/local/share/dart/cmake/DARTConfig.cmake:63 (include)
  /usr/local/share/dart/cmake/DARTConfig.cmake:123 (dart_traverse_components)
  /usr/local/share/dart/cmake/DARTConfig.cmake:171 (dart_package_init)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:124 (find_package)
  CMakeLists.txt:34 (find_package)

-- Found DART: /usr/local/include (Required is at least version "6") found components:  dart 
CMake Warning at /usr/local/lib/cmake/boost_program_options-1.71.0/libboost_program_options-variant-shared.cmake:59 (message):
  Target Boost::program_options already has an imported location
  '/usr/local/lib/libboost_program_options-mt.dylib', which will be
  overwritten with '/usr/local/lib/libboost_program_options.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_program_options-1.71.0/boost_program_options-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:159 (find_package)
  CMakeLists.txt:34 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_iostreams-1.71.0/libboost_iostreams-variant-shared.cmake:59 (message):
  Target Boost::iostreams already has an imported location
  '/usr/local/lib/libboost_iostreams-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_iostreams.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_iostreams-1.71.0/boost_iostreams-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
[... 51 lines cut ...]
--   Found libavformat, version 58.29.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.54.100
-- Checking for module 'libavutil'
--   Found libavutil, version 56.31.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.0
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Checking for module 'libzip'
--   Found libzip, version 1.5.2
CMake Error at cmake/FindGStreamer.cmake:54 (message):
  GStreamer not found
Call Stack (most recent call first):
  CMakeLists.txt:42 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/alexanderwu/Downloads/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /Users/alexanderwu/Downloads/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.15.4/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /Users/alexanderwu/Downloads/Firmware/Tools/sitl_gazebo && /usr/local/Cellar/cmake/3.15.4/bin/cmake -E touch /Users/alexanderwu/Downloads/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
(base) Alexanders-MacBook-Pro:Firmware alexanderwu$ 
`
Thanks a lot - any help would be greatly appreciated!

alexander-wu-aw, 2019-10-18:
Changed CMakeList.txt to not require gstreamer

closed 2019-10-18 by alexander-wu-aw

TSC21, 2019-10-18:
We have ongoing solutions for this:
https://github.com/PX4/sitl_gazebo/pull/353
https://github.com/PX4/sitl_gazebo/pull/354