PX4/PX4-Autopilot #13229: In the hold mode, the firmware 1.9.2 and 1.10 will have a throttle equal to 0

tonight123, 2019-10-18:
In the hold mode, the firmware 1.9.2 and 1.10 will have a throttle equal to 0, which changes quickly, resulting in a high degree of control. Highly controlled in firmware 1.8.2
1，firmware 1.9.2
[image]
[image]
[image]
2,firmware 1.10

[image]
3,firmware 1.8.2
[image]

julianoes, 2019-10-21:
Thanks for the issue. Can you please attache a link to the logfile? Or please share the "Velocity" plot.

MaEtUgR, 2019-10-21:
@tonight123 It looks from the pictures (a link to the log would give more insights) that your vehicle is well powered and can hover with ~26% thrust. Try adjusting the parameter `MPC_THR_HOVER` to `0.26` and it will likely solve the problem.

Explanation: The vehicle is by default assuming you need ~50% throttle to hover. If you give a lot less thrust but it's not falling it assumes you have contact with the ground and cuts the thrust to land but if you're in air it detects to be suddenly falling and catches again to not crash.

Future solutions for that reoccurring problem:
1. Estimate the hover thrust on the fly such that the user does not have to adjust. Here's the first draft for that: https://github.com/Auterion/Flight_Control_Prototyping_Scripts/pull/3
2. Improve the land detector to directly track the tracking of measured acceleration to thrust such that it's more robust on highly different vehicles.

tonight123, 2019-10-23:
@MaEtUgR in firmware v1.9.0 like PX4 V1.9.0 and Later: Attitude Control and Takeoff Issues #12484
takeoff： mpc_tko_ramp_t =0,it's going to take off quickly.
Starting counterclockwise after take-off, probably because rate i is too big, i is currently 0.2

closed 2019-10-24 by tonight123