PX4/PX4-Autopilot #13231: Land detector edge case in descend mode

julianoes, 2019-10-18:
**Describe the bug**
The land detector seems to detect ground contact in Descend mode (RC land failsafe without GPS).

1. `vertical_movement` to be `false`: https://github.com/PX4/Firmware/blob/95dc522b99688a08402ac6462ad28f5dfd338eda/src/modules/land_detector/MulticopterLandDetector.cpp#L154
2. `hit_ground` to be `true`: https://github.com/PX4/Firmware/blob/95dc522b99688a08402ac6462ad28f5dfd338eda/src/modules/land_detector/MulticopterLandDetector.cpp#L165

This happens because the z controller only reacts slowly to the new setpoint so the z velocity down is much less than the z velocity setpoint here:
https://github.com/PX4/Firmware/blob/95dc522b99688a08402ac6462ad28f5dfd338eda/src/modules/land_detector/MulticopterLandDetector.cpp#L151-L154

One observation of the descend case is that the velocity setpoint jumps to the land speed instead of gradually reaching it like in Land (with GPS) mode.

[image]

**To Reproduce**
This is hard to reproduce in SITL because the drone seems to follow the setpoint quicker.

**Expected behavior**
It should not detect ground contact until actually landed.

**Log Files and Screenshots**

This happened in a failsafe descend test:
https://review.px4.io/plot_app?log=3104d449-dd29-473f-9055-354984ff8afa

This does **not** happen in normal landing:
https://review.px4.io/plot_app?log=2ea6423f-b232-45c9-8c40-32e775118359

More info in https://github.com/PX4/Firmware/pull/13143#issuecomment-543682487.

FYI @bresch @MaEtUgR

MaEtUgR, 2019-10-22:
Related: https://github.com/PX4/Firmware/issues/13124, https://github.com/PX4/Firmware/issues/11683

LorenzMeier, 2021-01-02:
@MaEtUgR Can you think of a simple solution?

LorenzMeier, 2021-06-02:
This has been reworked since, closing.

closed 2021-06-02 by LorenzMeier