PX4/PX4-Autopilot #13232: jmavsim error OSX 10.14

srovani, 2019-10-19:
Hello all,

Very new here and to PX4/the Pixhawk.  I followed through the installation instructions provided  ([https://dev.px4.io/v1.9.0/en/](url)), but when attempting to start jmavsim, I get the following issues.  Unfortunately, at the same time, no secondary window opens in the background with the visuals for the sim.

`Samuels-MacBook-Pro:~ samuelrovani$ cd Firmware
Samuels-MacBook-Pro:Firmware samuelrovani$ make px4_sitl jmavsim
[0/1] cd /Users/samuelrovani/Firmware/build/px4_sitl_defa...mware /Users/samuelrovani/Firmware/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/samuelrovani/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /Users/samuelrovani/Firmware
build_path: /Users/samuelrovani/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/samuelrovani/Firmware/build/px4_sitl_default/bin/px4" "/Users/samuelrovani/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/Users/samuelrovani/Firmware"/test_data
INFO  [px4] Creating symlink /Users/samuelrovani/Firmware/ROMFS/px4fmu_common -> /Users/samuelrovani/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
You need to have Java 8 installed for macOS, for more info, see:
https://github.com/PX4/jMAVSim/issues/81
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CAL_BARO_PRIME: curr: 0 -> new: 6620172
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
ERROR [px4_work_queue] setting stack size for wq:hp_default failed (22)
ERROR [px4_work_queue] setting stack size for wq:att_pos_ctrl failed (22)
ERROR [px4_work_queue] setting stack size for wq:rate_ctrl failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-10-19/03_27_00.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh>  commander takeoff
ERROR [commander] rejecting takeoff, no position lock yet. Please retry..
Command 'commander' failed, returned 1.
pxh>`

Upon receiving these errors, I attempted to use Gazebo, but received the same issues as from [PX4/sitl_gazebo#353](url)  and [PX4/sitl_gazebo#354](url).  I recognize these are in-progress and should be resolved shorty.  I'm mostly wondering what I am doing wrong with jmavsim as a basic sim?

Thank you in advance!
Sam

julianoes, 2019-10-19:
Are you using Java 8?

```
You need to have Java 8 installed for macOS, for more info, see:
PX4/jMAVSim#81
```

srovani, 2019-10-19:
Hello Julianoes, thanks for the quick reply!

I do believe I am, as I installed Java 8 both from terminal, and using the 'typical' install off of the internet.  As before, I apologize for my inexperience with respect to terminal and how to use it.

To verify I had the proper installs, I used two methods, one from the System Preferences Java menu (pic below):
[image]

and one via terminal.  
[image]

Again, not sure if this is correct, but it seems like these may be conflicting as I have java 8 installed, but the open source java (OpenJKD build 13+33) installed in terminal?

Thanks again!

julianoes, 2019-10-21:
```
export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)
```

However, this should be done automatically:
https://github.com/PX4/Firmware/blob/f956bafa4eda7ceb298790ef0fff5540fd3c589f/Tools/jmavsim_run.sh#L65

julianoes, 2019-10-21:
Try:
```
rm -rf Tools/jMAVSim/out
export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)
make posix jmavsim
```

srovani, 2019-10-21:
Hello Julianoes,

Upon running "export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)" it appears to have changed the Java version to build 1.8.0_222-b10 which appeared to be promising, but still provided me with the same error in the end.

`Samuels-MacBook-Pro:~` samuelrovani$ export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)
Samuels-MacBook-Pro:~ samuelrovani$ java -version
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_222-b10)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.222-b10, mixed mode)
Samuels-MacBook-Pro:~ samuelrovani$ cd Firmware
Samuels-MacBook-Pro:Firmware samuelrovani$ make px4_sitl jmavsim
[0/1] cd /Users/samuelrovani/Firmware/build/px4_sitl_default/tm...ni/Firmware /Users/samuelrovani/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/samuelrovani/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /Users/samuelrovani/Firmware
build_path: /Users/samuelrovani/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/samuelrovani/Firmware/build/px4_sitl_default/bin/px4" "/Users/samuelrovani/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/Users/samuelrovani/Firmware"/test_data
INFO  [px4] Creating symlink /Users/samuelrovani/Firmware/ROMFS/px4fmu_common -> /Users/samuelrovani/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
You need to have Java 8 installed for macOS, for more info, see:
https://github.com/PX4/jMAVSim/issues/81
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CAL_BARO_PRIME: curr: 0 -> new: 6620172
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
[... 95 lines cut ...]
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
ERROR [px4_work_queue] setting stack size for wq:hp_default failed (22)
ERROR [px4_work_queue] setting stack size for wq:att_pos_ctrl failed (22)
ERROR [px4_work_queue] setting stack size for wq:rate_ctrl failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-10-21/17_10_43.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> commander takeoff
ERROR [commander] rejecting takeoff, no position lock yet. Please retry..
Command 'commander' failed, returned 1.
pxh> '

By the way, apologies for the way the code is showing up, I am using the "Insert Code" quotations, just seems to not be liking me.

Thanks again for the help!

srovani, 2019-10-21:
Out of curiosity, are the errors:

`ERROR [px4_work_queue] setting stack size for wq:hp_default failed (22)
ERROR [px4_work_queue] setting stack size for wq:att_pos_ctrl failed (22)
ERROR [px4_work_queue] setting stack size for wq:rate_ctrl failed (22)`

being caused by the lack of the proper version Java, or are these independent issues.

dagar, 2019-10-21:
@srovani those are unrelated, but it would be helpful if someone could fix them on MacOS.

It will be something platform difference that needs to be handled here. https://github.com/PX4/Firmware/blob/6a0f5249f825effc6a516596aea7ccc4b8c70cc2/platforms/common/px4_work_queue/WorkQueueManager.cpp#L211-L215

julianoes, 2019-10-22:
@srovani thanks for reporting back.

closed 2019-10-22 by julianoes

srovani, 2019-10-22:
@julianoes 
Apologies for any confusion, but as far as I can tell, this issue has not yet been resolved.  Upon running jMAVSim, no simulation window appears, and I am still unable to command the drone to takeoff or respond to any commands.

julianoes, 2019-10-25:
@srovani can you check with Oracle Java 8 instead of OpenJDK 8?

srovani, 2019-10-25:
As before, apologies for my lack of experience (this is my first major endeavor in terminal), but from the research I have done, it seems like Oracle Java 8 is no longer available on HomeBrew.  Everything I am finding online is to install JKD.

I am also wondering if jmavsim is necessary at this point.  I just would like to test the code that I am going to be writing for the aircraft/verify that it is set up properly.  If anyone has any other suggestions for this I am more than willing to consider them.

TSC21, 2019-10-25:
> As before, apologies for my lack of experience (this is my first major endeavor in terminal), but from the research I have done, it seems like Oracle Java 8 is no longer available on HomeBrew. [...]

Not trying to have you avoiding the issue, but why not give a try to Gazebo: https://dev.px4.io/v1.9.0/en/simulation/gazebo.html

srovani, 2019-10-25:
@TSC21  I did try Gazebo, unfortunately it was giving me the error that GStreamer was not found.  Not sure if this has been resolved, but I this error in [PX4/sitl_gazebo#353](https://github.com/PX4/sitl_gazebo/pull/353) and [PX4/sitl_gazebo#354](https://github.com/PX4/sitl_gazebo/pull/354).

srovani, 2019-10-25:
Edit: Should have probably posted the code in the previous message.  Here it is when I try to run Gazebo:
`Samuels-MacBook-Pro:~ samuelrovani$ cd Firmware
Samuels-MacBook-Pro:Firmware samuelrovani$ make px4_sitl gazebo_plane
[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
CMake Warning at /usr/local/lib/cmake/boost_system-1.71.0/libboost_system-variant-shared.cmake:59 (message):
  Target Boost::system already has an imported location
  '/usr/local/lib/libboost_system-mt.dylib', which will be overwritten with
  '/usr/local/lib/libboost_system.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_system-1.71.0/boost_system-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:33 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake:59 (message):
  Target Boost::filesystem already has an imported location
  '/usr/local/lib/libboost_filesystem-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_filesystem.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_filesystem-1.71.0/boost_filesystem-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:33 (find_package)

-- Found Boost: /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components:  system thread filesystem 
CMake Warning at /usr/local/lib/cmake/boost_regex-1.71.0/libboost_regex-variant-shared.cmake:59 (message):
  Target Boost::regex already has an imported location
  '/usr/local/lib/libboost_regex-mt.dylib', which will be overwritten with
  '/usr/local/lib/libboost_regex.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_regex-1.71.0/boost_regex-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  /usr/local/share/dart/cmake/DARTFindBoost.cmake:24 (find_package)
  /usr/local/share/dart/cmake/dart_dartComponent.cmake:6 (include)
  /usr/local/share/dart/cmake/DARTConfig.cmake:63 (include)
  /usr/local/share/dart/cmake/DARTConfig.cmake:123 (dart_traverse_components)
  /usr/local/share/dart/cmake/DARTConfig.cmake:171 (dart_package_init)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:124 (find_package)
  CMakeLists.txt:34 (find_package)

-- Found DART: /usr/local/include (Required is at least version "6") found components:  dart 
CMake Warning at /usr/local/lib/cmake/boost_program_options-1.71.0/libboost_program_options-variant-shared.cmake:59 (message):
  Target Boost::program_options already has an imported location
  '/usr/local/lib/libboost_program_options-mt.dylib', which will be
  overwritten with '/usr/local/lib/libboost_program_options.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_program_options-1.71.0/boost_program_options-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:159 (find_package)
  CMakeLists.txt:34 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_iostreams-1.71.0/libboost_iostreams-variant-shared.cmake:59 (message):
  Target Boost::iostreams already has an imported location
  '/usr/local/lib/libboost_iostreams-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_iostreams.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_iostreams-1.71.0/boost_iostreams-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:159 (find_package)
  CMakeLists.txt:34 (find_package)
[... 48 lines cut ...]
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 58.8.100
-- Checking for module 'libavformat'
--   Found libavformat, version 58.29.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.54.100
-- Checking for module 'libavutil'
--   Found libavutil, version 56.31.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.0
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Checking for module 'libzip'
--   Found libzip, version 1.5.2
CMake Error at cmake/FindGStreamer.cmake:54 (message):
  GStreamer not found
Call Stack (most recent call first):
  CMakeLists.txt:42 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/samuelrovani/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /Users/samuelrovani/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.15.4/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /Users/samuelrovani/Firmware/Tools/sitl_gazebo && /usr/local/Cellar/cmake/3.15.4/bin/cmake -E touch /Users/samuelrovani/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1`

ikuraj, 2020-01-10:
Just to confirm that installing Oracle JVM solved the issue for me (as this was suggested and not test/confirmed).

julianoes, 2020-02-12:
For anyone seeing this later, check out: https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting