PX4/PX4-Autopilot #13245: Can't load node in gazebo

roseyangyu, 2019-10-22:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

closed 2019-10-23 by roseyangyu

Jaeyoung-Lim, 2019-10-23:
@roseyangyu You might need to translate the description file to sdf

roseyangyu, 2019-10-29:
> @roseyangyu You might need to translate the description file to sdf

I have run "roslaunch px4 posix_sitl.launch" follow [this](https://dev.px4.io/v1.9.0/zh/simulation/ros_interface.html).I solve that