PX4/PX4-Autopilot #13250: Failure to build with cm8jl65 driver

jordancoult, 2019-10-23:
**Bug Description**
When trying to build for px4_fmu-v2_default with the cm8jl65 driver added to default.cmake, I get an error which prevents the build from finishing (see screenshots). Everything builds fine when I leave default.cmake as-is.

**To Reproduce**
1. Clone master
2. Add the line `distance_sensor/cm8jl65` to the drivers section of /Firmware/boards/px4/fmu-v2/default.cmake
3. Navigate to /Firmware
4. Run `make px4_fmu-v2_default`

**Additional Info**
I set up the development environment by following the official instructions on a clean Ubuntu 18.04 VM. 

**Full Console Output**
[image]
[image]

mcsauder, 2019-10-23:
Hi @jordancoult , I was able to reproduce your issue, thanks for reporting it!  Let me know if PR #13258 solves your issue!

jordancoult, 2019-10-23:
Hi @mcsauder, thanks for the help. I tried your source branch and the build threw a different error: 

"Error: PX4 version too low, expected at least v1.9.0
Check the git tag (current tag: 'v1.7.3')"

It seems that your branch has outdated tags. To fix this, I cloned fresh from master and applied your changes and the build was successful.

mcsauder, 2019-10-24:
Hi @jordancoult , I have fetched `PX4/Firmware` 's  upstream tags and pushed them to my fork.  You should no longer see the tag issue.  Thanks for validating the patch!

mcsauder, 2019-10-24:
Closing. Fixed with #13258!

closed 2019-10-24 by mcsauder