PX4/PX4-Autopilot #13252: Throttle drops to min and pitch down commanded after front transition

semihyildirim, 2019-10-23:
I have a heavy vtol. I had lots of flights but in everyflight, the autopilot drops the throttle to minimum value(FW_THR_MIN=%40), and maximum negative pitch commaned after front transition occurs and the plane switch to fw mode. In last flight, there was lots of wind and the plane could not compansate these error and I had a big crash. 
Why throttle dropping and pitch down commanded by autopilot? 
Do not mention about airspeed or gain errors on our flight log. 
Say why the autopilot causes a stall by doing throttle drop? 
If the autopilot had not been dropped the throttle to min the plane had not been crashed. 
My plane has electric setup. On pixhawk document says that throttle min should be set to 0 in electric setups. If the throttle min had been set to 0, the autopilot drops to throttle to zero. This is more dangerous thing then %40 throttle min value.

Flight log: https://review.px4.io/plot_app?log=e4f597d2-f90f-443e-996c-6c6bc3b8b799

bresch, 2019-10-23:
FYI @sfuhrer  @RomanBapst

semihyildirim, 2019-10-23:
If autopilot try to connect altitude so give maximum pitch after transition, because of minimum pitch commanded throttle must be lowest value. According to this idea, minimum throttle(%40) may be overshoot for achieving these strategy. 

In previous flight, transition airspeed is lower than this flight so the plane was more slow. The autopilot could be achieve in lower speeds but in this flight the plane had more speed because of higher transition airspeed.

semihyildirim, 2019-10-23:
The following line of codes could be affect these behaviour? 

https://github.com/PX4/Firmware/blob/26364d44c91672d101349ff5825358017f7db3df/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp

Line number between: 1300- 1316

RomanBapst, 2019-10-25:
@semihyildirim Thanks for reporting this!
I need to investigate why the throttle is kept at minimum for such a long time after transition. I would expect it to go to cruise throttle, not min throttle.
Anyway, I think there is something else not quite right with this setup.
After the transition the aircraft pitches down but the setpoint commands pitch up. It's weird that even when airspeed increases to high values that there is no pitch response at all. 
Your trim airspeed is set to 19m/s but even at 26 m/s I see no response from the pitch loop, even though the controller demands pitch up.
Either your vehicle is very nose heavy or your control surfaces are very ineffective.

I will follow up on the throttle issue, can you follow up on the no pitch response issue?

RomanBapst, 2019-10-25:
[image]

semihyildirim, 2019-10-25:
Look at the ground speed and see highwr speeds.

semihyildirim, 2019-10-25:
I want to learn why min pitch down and min throttle commanded by pixhawk, you say that why the plane would not able to pitch up? 
I think the plane want to correct altitude error but do it agressively. Is it true or not?

RomanBapst, 2019-10-25:
@semihyildirim Ground speed does not matter, airspeed is important for your plane to fly.
Again, form what I see in the logs the controller demands pitch up, check it!
Why is the plane not able to follow the setpoint? Maybe your pitch controller is not tuned properly.

semihyildirim, 2019-10-25:
Sorry, issue closed wrongly.

semihyildirim, 2019-10-25:
In addition, the pitch controller affects pitch down and pitch up movements. The autopilot give pitch down command within 0sn but it give pitch up command within 3-4sn. Why? If I increase pitch controller, pitch down commaded more agressively?  Is it right?

closed 2019-11-11 by pifordi