PX4/PX4-Autopilot #13279: Position estimation error when the autopilot is on the ground?

xdwgood, 2019-10-27:
Branch: Latest master

Issue:
I placed the autopilot at rest, select the position mode, the drone local.x/y position can not be locked

test way:
[image]

Note: Raw gps latitude and longitude data
[image]

In the case of positional drift, should we let the drone enter the position mode? Or what am I doing wrong?

log link:https://review.px4.io/plot_app?log=9cbdc1d9-a9aa-4822-af7e-7cdd04e8a339

FYI: @huangwen0907  :smiley:

mcsauder, 2019-10-27:
Hi @pandacoolcool, are you in between tall buildings? If yes, allow 30 minutes or more with your gps receiver on the ground. You will get faster lock if you raise it off the ground by half a meter or more.

xdwgood, 2019-10-27:
Hi @mcsauder  There are no more than 10 meters of buildings nearby. I deliberately chose a relatively empty location for testing.

xdwgood, 2019-10-27:
I forgot to try to arm, I don't know if It will check the position drift before arm....but I think we should not enter the position mode when the position drifts...unless the drone starts on the moving car..  :smiley:

mcsauder, 2019-10-27:
If this is the first time this unit has gotten lock in this area, or if you have had the unit turned on indoors previously where little or no gps signal was available, give it a ridiculous amount of time for the first lock!

xdwgood, 2019-10-27:
> If this is the first time this unit has gotten lock in this area, or if you have had the unit turned on indoors previously where little or no gps signal was available, give it a ridiculous amount of time for the first lock!

@mcsauder  sorry,I don't quite understand what you mean...In fact, the gps prompt has completed 3d lock, and there are enough satellites... the autopilot also allows me to enter the position mode.... What is the lock you said?

huangwen0907, 2019-10-28:
@xdwgood  Pls see the picture below, you can find the innovation of horizontal is about 0.2,   I think this is the normal value, and the black line show that you have a valid position, so the localization is good as the ekf's sides. 
However you can find that the position is always drift, so I think the reason is that: you do not have a high quality gps module or not locate a good gps place. Many gps module in the market is not good as YUNEEC's.
[image]

I also diff the velocity, we can find that the gps's velocity is matched with the estimation value.   
[image]

BTW: 
You can find the gps quality, I think the eph and epv is not very good.
[image]

At last, I think the best way is change a gps module or optimize the ekf's measurement noise, if I, I will do the second way, you can get a power on time log and do the replay work.

closed 2019-10-28 by xdwgood