PX4/PX4-Autopilot #13280: LL40LS driver cause crash

hyonlim, 2019-10-27:
**Describe the bug**
With current master & Pixhawk4 combo, when LL40LS app started command line interface stopped. If LL40LS app is set to started during boot, nsh command is not working.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash master
2. Connect LL40LS to i2c
3. Set SENS_EN_LL40LS Enabled
4. See console output (via console cable)

**Expected behavior**
When type a command on nsh, command should work.

**Log Files and Screenshots**
Attached.
[image]

dagar, 2019-10-27:
@mcsauder have you successfully used the LL40LS on i2c recently?

dagar, 2019-10-27:
@hyonlim would you be able to get the full boot output from the beginning?

dagar, 2019-10-27:
This is a bit of a leap, but one possibility is we may need to increase `CONFIG_MAX_TASKS`. 

https://github.com/PX4/Firmware/compare/master...dagar:pr-task_max?expand=1

hyonlim, 2019-10-27:
https://github.com/PX4/Firmware/pull/13285 solves this problem. Thanks!

closed 2019-10-27 by hyonlim